espressif/idf-eclipse-plugin #1050: Project Launch Bar Missing

cjdg, 2024-09-16:
**Describe the bug**
There is no Project Launch Bar on Eclipse.

**To Reproduce**
Install Espressif IDF plugin
Create a new project
No toolbar

**Expected behavior**
Its supposed to have a toolbar for building and running the project.

**Screenshots**
[image]

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)

**Eclipse Error log:** 

[Error.log](https://github.com/user-attachments/files/17016236/Error.log)

sigmaaa, 2024-09-17:
Hi @cjdg,
Please take a look if you have the launch bar enabled in preferences:
1. In eclipse go to Window -> Preferences -> Run/Debug -> Launching -> Launch Bar
[image]

cjdg, 2024-09-18:
[image]

cjdg, 2024-09-18:
[image]

sigmaaa, 2024-09-18:
Hi @cjdg,
I see Launching is unclicked on your screenshot. The "Launch Bar" option must appear after you click on "Launching"

cjdg, 2024-09-18:
[image]

sigmaaa, 2024-09-19:
Hi @cjdg,

could you check if the LaunchBar is installed in the CDT plugin? Please try the following:

- Go to Install New Software
- Select the CDT update site, for example - https://download.eclipse.org/tools/cdt/releases/latest
- Install the Launchbar plugin if it's missing. Please find the screenshot below.
[image]

cjdg, 2024-09-19:
that was!!!!, well, now i have the toolbar and works fine, thanks, maybe it has to be writen as a requirement for the plugin, or extend the documentation ?

closed 2024-09-19 by cjdg