espressif/idf-eclipse-plugin #1071: ESP-IDF Terminal in Espressif IDE not going to the project path

sanderdb01, 2024-10-25:
I am doing this on a mac M1, and according to the IDE getting started (https://github.com/espressif/idf-eclipse-plugin?tab=readme-ov-file#idfterminal), if I want to open the ESP-IDF terminal to the project directory I have to select the project and then open the esp-idf terminal. However, it will not do that. It will only ever open to the esp-idf library directory (IDF_PATH). I also tried right clicking the project and selecting "show in local terminal", but there are no selectable options. I then tried installing a 3rd party terminal (which then gave me an option to open in a local terminal), but it was not an esp-idf terminal, so none of the idf.py commands could be recognized.

How do I make it so that the esp-idf terminal opens to the project folder on mac?

sigmaaa, 2024-10-29:
Hi @sanderdb01,

here is an example how I'm opening esp-idf terminal in the project folder:

https://github.com/user-attachments/assets/25b37db8-2790-494b-a57a-a568b9ba661a

Let me know if that is how you are trying to open it and if you still having an issue with it. Thank you

sanderdb01, 2024-10-29:
ahh yes, that worked. I was clicking the terminal icon down at the bottom of the screen, not the top. That actually made a difference. Thanks a bunch

closed 2024-10-29 by sanderdb01

sigmaaa, 2024-10-30:
Hi @sanderdb01,

I'm glad it's working for you now. We will add more options to open the esp-idf terminal to make it more user-friendly. I agree that the current implementation may be confusing. Thank you