espressif/idf-eclipse-plugin #1072: Espressif-IDE Code Formatting not working

sanderdb01, 2024-10-29:
When I open a c file in the editor, none of the shortcuts or code editing works. When I click "Source" on the menu bar, everything in grayed out. I tried resetting my perspective to C/C++, but that still did not solve it. I am on a Mac if that helps. Thanks.

sigmaaa, 2024-10-30:
Hi @sanderdb01,

some options don't work with the new LSP editor that we are using. You can still use some Source commands with the right click inside the editor. For the formatting you can try the default key combination `cmd+shift+F". You can also set to auto-trigger the formatting with the save action. To do this go to the `Preferences -> C/C++ -> Editor(LSP) -> Format source code`
[image]

Note that the formatter settings could be changed by editing the `.clang-format` file inside the project. https://github.com/eclipse-cdt/cdt-lsp?tab=readme-ov-file#save-actions-using-clang-format

sanderdb01, 2024-10-31:
got you, thank you for the response. Is there a way to auto comment? Usually on other IDEs I can do cmd/ctrl+ '/' to comment out a line, but that does not work here. Thanks.

sigmaaa, 2024-10-31:
Hi @sanderdb01,

the default combination for commenting out the line in the LSP editor is `ctrl + 7`, but `cmd+shift+/` should also work. I was able to reproduce this issue and will take a closer look into it. It's also possible to rebind this combination. To do this:
- Go to the `Preferences -> General -> keys -> look for `Toggle Line Comment` in the TM4E configuration:
[image]

Thank you