espressif/idf-eclipse-plugin #1073: ESP-IDF Manger - Installing IDF v5.1.5 (any other version have the same behavor)

huwi, 2024-10-31:
**Describe the bug**

Thank you for your great work.

After installing eclipse and added esp idf plugin, the es-idf manager cant install the idf environment. 
-> Espressif->ESP-IDF Manger->Add ESP-IDF
The process is running. at the end a message pops up:

Problem Occurred
Cannot invoke "String indexOf(int) because "value" is null
=> Details
Cannot invoke "String.indexOf(int)" because "value" is null
Cannot invoke "String.indexOf(int)" because "value" is null

**Espressif-IDE Product Information: 
Eclipse Plugin 3.1.0.2024.100707 (From Marketplace)
ESP-IDF v1.5.1, ESP-IDF v5.3.1, ....

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log 
[image]
[image]
[image]
[image]

Kind Regards, Hubert

alirana01, 2024-10-31:
@huwi thanks for reporting this issue.

Can I please ask you to download the esp-idf required version via the IDE. This is a known issue but it was only reported for the master branch. We have the fix for that in the upcoming release. For other versions it should not report this error. Also please export and attach the logs so we can investigate the details.

alirana01, 2024-10-31:
@huwi another thing I just noticed is that you are specifying the python from virtual env folder. The python needs to be the system python not from the virtual env directory. Please try to use the system python when installing tools. The virtual env python is picked up by the IDE itself when the tools are installed.

apalazzi, 2024-11-19:
Hi,
I'm having the same issue. Clean Eclipse install, then installed the IDF plugin via marketplace. The error occurs when installing the toolchain but also when trying to use the tools installed separately as explained here: https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/get-started/linux-macos-setup.html

In my case python is the system python.

alirana01, 2024-11-20:
@apalazzi the issue will be fixed with the upcoming release.

mmrein, 2024-11-21:
@alirana01 ETA on upcoming release or is there some workaround?

Tried resatrting and using system python path, issue still exists.

apalazzi, 2024-11-21:
@mmrein I haven't tested it deeply, but it seems to work on the latest nightly builds: https://github.com/espressif/idf-eclipse-plugin?tab=readme-ov-file#installing-idf-plugin-using-update-site-url

mmrein, 2024-11-21:
@apalazzi Ok, seems to be working, thank you

lbthomsen, 2024-12-03:
It does NOT work for me with the latest nightly build.

[image]

[image]

Same as before.

komcconnell, 2024-12-05:
This still failed for me as of last night.  M3 MacBook Pro running Mac OS X 14.6.1 (Sonoma).

StudlySpud, 2025-01-06:
Hi, this is still failing for me.  I notice that 5.4 was released just yesterday!  

I am running Ubuntu, I have tried through both the Espressif IDE and the Eclipse IDE to add ESP-IDF by automatically downloading v5.4.  
I have also seperately tried to download and setup the tools as described here, this is successful.  
https://docs.espressif.com/projects/esp-idf/en/v5.4/esp32/get-started/linux-macos-setup.html

But adding the existing ESP-IDF directory from file system fails in the same way.  

My symptoms are same as shown by others exits after this line;
`Executing /home/spudadmin/.espressif/python_env/idf5.4_py3.10_env/bin/python /home/spudadmin/esp/esp-idf/tools/idf.py --list-targets`
With an error dialog box showing;
Cannot invoke "String indexOf(int) because "value" is null.

The only extra clue I haven't seen reported in this or other issues is that earlier in the log, it seems to persistently download, check, fail, then delete the tool qemu-xtensa

```
Installing qemu-xtensa@esp_develop_9.0.0_20240606
Using GitHub assets mirror for URL: https://github.com/espressif/qemu/releases/download/esp-develop-9.0.0-20240606/qemu-xtensa-softmmu-esp_develop_9.0.0_20240606-x86_64-linux-gnu.tar.xz => https://dl.espressif.com/github_assets/espressif/qemu/releases/download/esp-develop-9.0.0-20240606/qemu-xtensa-softmmu-esp_develop_9.0.0_20240606-x86_64-linux-gnu.tar.xz
file qemu-xtensa-softmmu-esp_develop_9.0.0_20240606-x86_64-linux-gnu.tar.xz is already downloaded
Extracting /home/spudadmin/.espressif/dist/qemu-xtensa-softmmu-esp_develop_9.0.0_20240606-x86_64-linux-gnu.tar.xz to /home/spudadmin/.espressif/tools/qemu-xtensa/esp_develop_9.0.0_20240606
ERROR: tool qemu-xtensa version esp_develop_9.0.0_20240606 is installed, but getting error: non-zero exit code (127) with message: /home/spudadmin/.espressif/tools/qemu-xtensa/esp_develop_9.0.0_20240606/qemu/bin/qemu-system-xtensa: error while loading shared libraries: libSDL2-2.0.so.0: cannot open shared object file: No such file or directory

ERROR: Failed to check the tool while installed. Removing directory /home/spudadmin/.espressif/tools/qemu-xtensa/esp_develop_9.0.0_20240606
```

petzah, 2025-01-07:
Problem seems to be with version string is "v5.4-dirty":

```
~/.espressif/python_env/idf5.4_py3.11_env/bin/python /esp-idf-v5.4/tools/idf.py --version
ESP-IDF v5.4-dirty
```

It looks like someone forgot to update/remove .git directory from the esp-idf-v5.4.zip file ;) and therefore version returned by idf.py is not what java code expects.

Removing the .git directory and then manually pointing to the extracted path seems to workarounds the issue as it falls back to read version from the source:

```
~/.espressif/python_env/idf5.4_py3.11_env/bin/python /esp-idf-v5.4/tools/idf.py --version
fatal: not a git repository: '/esp-idf-v5.4/.git'
WARNING: Git version unavailable, reading from source
ESP-IDF v5.4.0
```

alirana01, 2025-01-22:
This issue can also happen if your system environment variables contain the variables for esp-idf.

Please refer to the discussion on this issue I am closing this from here
https://github.com/espressif/idf-eclipse-plugin/issues/1105

closed 2025-01-22 by alirana01