espressif/idf-eclipse-plugin #1074: Code completion: "Error while computing completion"

zamek42, 2024-10-31:
**Describe the bug**
Code completion does not work, I see " Error while computing completion" 
**To Reproduce**
Steps to reproduce the behavior:
1. open an Esp 32 IDF project,
2. open a .c file
3. go to a function and pres any key (l) and Ctrl+Space 
4. See error:  "Error while computing completion"

**Expected behavior**
I want to see the completion list including function names, variables and code templates. 

**Screenshots**

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)
[product_info.txt](https://github.com/user-attachments/files/17586615/product_info.txt)

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)
[code_completion_error.log](https://github.com/user-attachments/files/17586567/code_completion_error.log)
Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

sigmaaa, 2024-11-01:
Hi @zamek42,

Could you please check if the LSP editor settings are configured correctly? To do this go to `Preferences -> C/C++ -> Editor(LSP) ->clangd`. My settings for esp32 look like this:
[image]

If you are using the latest ESP-IDF IDE or plugin, these settings should be configured automatically after installing the tools, setting the target, and building the project. Here are some specifics:

- **Path Options:** This should be set up automatically after tool installation. If it's not, try clicking Restore Defaults on the clangd preference page.
- **Drivers:** This setting depends on the target selected in the launch bar. If it’s empty, try reselecting the target; for example, select ESP32-S2, then switch back to ESP32.
- **Additional** Settings: These options are used for source navigation and are configured automatically after building the project.

Let me know if that helped. Thanks

zamek42, 2024-11-01:
Hi Sigmaaa, 

Thx for your replay. When I click to clangd settings I've got an error: "The currently displayed page contains invalid values". 
And then when I click somewhere else and click back it displayed but every fields are empty. And when I fill everything as you suggested, and save it, it seems  it does not save it. 

thx,
Zamek

Pastovky, 2024-11-07:
Hello, I encountered the same problem while using Espressif-ID v3.0.0. Upon reinstalling, I noticed that I do not have the "esp-clang" folder at the following path: Espressif\tools\esp-clang. However, in the Esp-IDE (Preferences -> C/C++ -> Editor (LSP) -> clangd), the settings are automatically configured to the same paths as yours, even though one of the folders is missing. I added the missing folder with the contents, reset the settings in ESP-IDE to the default, and manually re-entered them. After that, the problem was resolved. I restarted the system and the program, and everything was fine, but after the first compilation of any project, the problem returned.

GrantMTG, 2024-11-09:
New install this week (Win11). I had the same problem and your fix worked. In my case the path just said clangd and nothing else.  Everything else was correct. I browsed to the clangd.exe and that worked. Thanks.

AndriiFilippov, 2024-11-15:
@GrantMTG hi ! 
Could you please provide more details about your setup, because I have tried latest IDF plugin v3.1.0 with both Eclipse 24-06 / Eclipse 24-09 and couldn't reproduce this issue. Also tried latest IDE v3.1.0, also works fine.
We do know about this issue in Espressif-IDE v3.0.0, but from your message it isn't clear if you used 3.0.0 or 3.1.0.
You can download latest one here: [Espressif-IDE v3.1.0](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.1.0).

GrantMTG, 2024-11-15:
Hi. I have two installations. I'm at work now and everything is OK here. v3.1.0 (Win10). I will check at home (Win11) and see if it's fixed or not. It's been a few crazy days so I can't remember for sure. I know at work I have none of the VS Code/PlatformIO/Arduino stuff. Very clean install and working good.

zamek42, 2024-11-16:
Hi, I've downloaded the latest 3.1.0 as you suggested, bu the no changes :(. 
I have a Debian Linux Bookworm and java:
openjdk version "17.0.13" 2024-10-15
OpenJDK Runtime Environment (build 17.0.13+11-Debian-2deb12u1)
OpenJDK 64-Bit Server VM (build 17.0.13+11-Debian-2deb12u1, mixed mode, sharing)
![PXL_20241116_081607679](https://github.co
[image]
m/user-attachments/assets/7a55ba82-d10c-4240-9c2c-abfbd372647e)
[image]

I tried eclipse (Version: 2024-09 (4.33.0) Build id: 20240905-0614) and I installed  plugin. Code completion works on it, but it does not contains the editor templates (Preferences/C/C++/Editor/Content Assist/Templates. Clangd in this eclipse is ok. 

thx,
Zamek

GrantMTG, 2024-11-17:
@AndriiFilippov
Mine is working OK now. Honestly I can't remember what I did, but it was nothing too sophisticated. It may even have been closing eclipse and reopening it and/or rebooting.

[image]

GB

LAPSyLAB, 2024-12-22:
> Hi Sigmaaa,
>
> Thx for your replay. When I click to clangd settings I've got an error: "The currently displayed page contains invalid values". And then when I click somewhere else and click back it [...]

Hello,

I get exactly the same behaviour on 
Version: 2024-12 (4.34.0)
Build id: 20241128-0757

I can find the files on paths, but settings are always erased....

sigmaaa, 2025-03-07:
Hi @ua100k, @Radot1, @LAPSyLAB 

Could you please try Espressif-IDE from [this link](https://github.com/espressif/idf-eclipse-plugin/actions/runs/13673556862?pr=1171) and check if you encounter the same issue on this PR? Thank you

zamek42, 2025-03-07:
Hi All,
Seems it works well! :) Thank you. 
 There is only one problem remains, my code templates does not works. 
I made a template in Windows/Preferences/Editor/Templates: new:

[image]

 but it does not exists in loge Ctrl+space list. 
thx,
Zamek

sigmaaa, 2025-03-07:
Hi @zamek42,

Unfortunately, code templates are not yet supported in the LSP CDT editor: https://github.com/eclipse-cdt/cdt-lsp/issues/333.

zamek42, 2025-03-07:
Hi @sigmaaa 

thx a lot, I can leave without code templates.

Radot1, 2025-03-08:
> Hi [@ua100k](https://github.com/ua100k), [@Radot1](https://github.com/Radot1), [@LAPSyLAB](https://github.com/LAPSyLAB)
>
> Could you please try Espressif-IDE from [this [...]

Hi @sigmaaa, it worked. Cheers