espressif/idf-eclipse-plugin #1075: ESP-IDF Manager: Window "broken"

FabITA1, 2024-11-06:
**Describe the bug**
After installing some ESP-IDF from disk, the ESP-IDF Manager windows "broke" (see screenshot). Even closing and reopening the IDE, the problem persists. I also try to uninstall and reinstall the ESP-IDE, but the issue comes back again.

**To Reproduce**
I don't know if it's systematic. In a fresh Espressif-IDE installing, I have added IDF 4.4.4 and then 5.1. After that, the IDF Mangager appears like the screenshot.

**Expected behavior**
The ESP-IDF Manager window still works.

**Screenshots**
[image]

**Espressif-IDE Product Information:**
Version: 3.1.0
Build id: 20241007-0706

**Eclipse Error log:** 
[image]

kolipakakondal, 2024-11-07:
Hi @FabITA1 , for some reason, the ESP-IDF version was set to null, which caused this issue.

Could you delete the **tool_set_config.json** file from your workspace, then re-launch the ESP-IDF Manager and add the ESP-IDF version by selecting the "Using an existing ESP-IDF option..."

> /Users/myusername/myworkspacename/tool_set_config.json

Please note that Espressif-IDE 3.1.0 only supports ESP-IDF 5.0 and higher. In your case, I see you’re using ESP-IDF 4.4.4. If you plan to use ESP-IDF 4.4.4, please use [Espressif-IDE 2.12.1](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.12.1).

alirana01, 2024-11-08:
@FabITA1 I agree this can be an issue we are working on a fix for this and will close this issue once that is in the latest master for plugin in the mean time I can recommend a workaround to keep you going.

The file you shared tool_set_config.json is fine in terms of the id number. The issue however is in the envVars for the later ESP-IDF version 5.1 that you have installed.
You can try to update the json file and add the "ESP_IDF_VERSION" : "5.1" manually to it as show in the image below. Then please restart the IDE in the same workspace and set the required idf version to active.

[image]

FabITA1, 2024-11-08:
Hi @alirana01 thanks for your reply. I followed your suggestion by adding the missing line, but the issue persists. 
For now, I installed only the 5.3.1 what I need at the moment.

atreistm, 2024-12-12:
I am having a similar issue. I deleted the .json file and the Manager now opens properly. However, the ESP-IDF version field is not populated .

ntremble-sst, 2024-12-16:
I am also seeing this same issue. I have been using various IDF versions above 5.0 with no issue, but as soon as I install IDF 5.3 or higher this issue appears: all the IDF versions that were already showing in the ESP-IDF Manager disappear, and the value of IDF_PATH appears to default to esp_idf_master, which breaks my builds as they expect 5.2.2 or less.

The reason this issue appeared in the first place was that I installed IDF master in order to compile for the ESP32-P4.

I can still compile from the command line; it is only Eclipse + Espressif Plugin which are broken.

ntremble-sst, 2024-12-17:
To add to my previous comment, with only IDF 5.1.2 and IDF 5.2.2 in the IDF manager they were showing normally. I then closed my project and exited from Eclipse, then restarted Eclipse. It presented a warning as shown below, seemingly indicating it was looking for a file esp_idf.json.json (which does not exist) rather than tool_set_config.json (which does exist in the workspace).

[image]

It appears the plugin has incorrectly named the config file, cannot find it, then uses esp_idf_master as the default, which messes everything up.

If I say 'use old path' at this point then all is good, the IDF manager functions as expected. Now, if I install IDF 5.3 as well, then installation proceeds normally, but when complete the IDF manager is blank. If I close the manager tab at this point and reopen I will get the broken manager screen as is the subject of this issue.

Comparing the files before and after I see there are now two entries for IDF 5.2.2 and an entry for IDF 5.3. See attached files.

[tool_set_config-broken.json](https://github.com/user-attachments/files/18158864/tool_set_config-broken.json)
[tool_set_config-working.json](https://github.com/user-attachments/files/18158883/tool_set_config-working.json)

The only way I have found to recover is to replace the broken tool_set_config.json in the workspace. This allows the IDF manager to function correctly, but of course I do not have IDF 5.3 available, so I cannot reasonably develop for the ESP32-P4 (every time I habitually hit build in the IDE it clobbers my build directory, so I need to fullclean and build from the cli.

ntremble-sst, 2024-12-17:
I managed to construct a working toot_set_config.json file which allows me to build for ESP32-S3 with IDF 5.2.2, or for ESP32-P4 with IDF 5.3, in both cases within Eclipse, and with ESP-IDF Manager functional. See attached.

The only real difference I noticed was that the working file specifies "idfVersion": "5.3" and "ESP_IDF_VERSION": "5.3" whereas the broken file does not. That may or may not be relevant, I did not test it further once I had a working solution.

[tool_set_config.json](https://github.com/user-attachments/files/18159649/tool_set_config.json)

[image]

flodmotorgrodan, 2025-02-18:
This problem may be related to [Unable to install ESP-IDF 5.3.2 via ESP-IDF Manager in Elclipse (IEP-1395)](https://github.com/espressif/idf-eclipse-plugin/issues/1105#issuecomment-2666747708)

Where idf_tools.py does not export the path to the ESP-IDF virtual python environment.

ntremble-sst, 2025-02-18:
> This problem may be related to [Unable to install ESP-IDF 5.3.2 via ESP-IDF Manager in Elclipse (IEP-1395)](https://github.com/espressif/idf-eclipse-plugin/issues/1105#issuecomment-2666747708)
> 
> Where idf_tools.py does not export the path to the ESP-IDF virtual python environment.

Perhaps, although even in the presence of this error in the Eclipse IDE I am able to use the CLI to fullclean/reconfigure/build without issue, so I don't see any sign of a configuration error at that level.

Since I have worked out how to manually edit the tool_set_config.json to give me a working setup it has been working perfectly for me in either CLI or Eclipse. Refer my earlier comment: https://github.com/espressif/idf-eclipse-plugin/issues/1075#issuecomment-2547432694

So from my point of view this is a simple failure by the Espressif plugin during construction of the tools_config.json.