espressif/idf-eclipse-plugin #1078: Computing proposales not working

jh-elec, 2024-11-08:
**Describe the bug**
autocomplete not worked 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Espressif-IDE Product Information:**
Operating System: windows 10
Java Runtime Version: 21.0.5+11-LTS
Eclipse Version: 4.33.0.v20240903-0240
Eclipse CDT Version: 11.6.1.202406241230
IDF Eclipse Plugin Version: 3.1.0.202410070705
ESP-IDF v5.3.1-dirty
Python set for IDF_PYTHON_ENV: Python 3.11.9

**Eclipse Error log:** 
[err.log](https://github.com/user-attachments/files/17676515/err.log)

sigmaaa, 2024-11-12:
Hi @jh-elec,

I've noticed that the launch bar is missing. It's necessary for building the project with the correct target and configuration.
Could you check if the LaunchBar is installed in the CDT plugin? Please try the following:

- Go to Install New Software
- Select the CDT update site, for example - https://download.eclipse.org/tools/cdt/releases/latest
- Install the Launchbar plugin if it's missing. Please find the screenshot below.

[image]

If it's installed, but the launch bar is still missing, check the preferences:
`In eclipse go to Window -> Preferences -> Run/Debug -> Launching -> Launch Bar`
[image].

After turning on the launch bar, selecting the target, and building the project, the LSP editor options should be set automatically. Please check this https://github.com/espressif/idf-eclipse-plugin/issues/1074#issuecomment-2451564754 to find more details. Thanks

sigmaaa, 2024-11-12:
Hi @jh-elec,

> "Initializing langauge Server C/C++" and remains at 28%.
Is that normal?

No, usually this process is completed without problems. Could you please share the error log during that process hangs?
Also, could you please check if you are having the same issue on the Espressif-IDE https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-3.1.0-win32.win32.x86_64.zip. It's lightweight and contains all the prerequisite plugins. You have only select the path to the existing esp-idf via tools manager after opening it for the first time. Thanks

jh-elec, 2024-11-13:
I tried it with the other IDE and the result is exactly the same. Init. Langauge Server stalls at 28%.

[err.log](https://github.com/user-attachments/files/17735267/err.log)

sigmaaa, 2024-12-02:
Hi @jh-elec,

I'm sorry for not getting back to you sooner. We weren't able to reproduce this problem until today. The problem you are facing is due to a space in the build folder path
[image]

Here is an Espressif-IDE build with a fix https://github.com/espressif/idf-eclipse-plugin/actions/runs/12119729394/artifacts/2261424278
or you can use the update site https://github.com/espressif/idf-eclipse-plugin/actions/runs/12119729394/artifacts/2261423680

jh-elec, 2024-12-04:
Now i have used your fix (Here is an Espressif-IDE build with a fix https://github.com/espressif/idf-eclipse-plugin/actions/runs/12119729394/artifacts/2261424278)
But its same problem^^

sigmaaa, 2024-12-04:
Hi @jh-elec,

could you please verify if the build path is enclosed in quotation marks after the build? 
[image]

jh-elec, 2024-12-06:
No, it wasn't in quotation marks. I've now done it manually and lo and behold it works ;)

closed 2024-12-17 by kolipakakondal