espressif/idf-eclipse-plugin #1080: ESP-IDF install corrupted

goofy2k, 2024-11-16:
My ESP-IDF install (Espressif-IDE Eclipse plugin) got corrupted for some reason.  The projects in my workspace were not visible anymore. The project files were still present on my storage, so I decided to import them again with File > Import > Espressif > Existing IDF project. Then  the projects appeared again in my Project Explorer.

When I try to compile a project again I get the following responses:

- by right-clicking it and giving Clean Project:  
```
  CMakeFiles not found. Assuming clean.
 ```
- by right-clicking it and giving Build Project:

```
-- Checking Python dependencies...
ERROR: C:\Users\fredv\.espressif\espidf.constraints.v5.3.txt doesn't exist. Perhaps you've forgotten to run the install scripts. Please check the installation guide for more information.
CMake Error at C:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/build.cmake:365 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/build.cmake:502 (__build_check_python)
  C:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Espressif/workspace/TFT_JC2432S028/scan/build/CMakeFiles/CMakeOutput.log".
cmake --build . -- -v
ninja: error: loading 'build.ninja': The system cannot find the file specified.
 ```

Here is the info that I get when I give Espressif > Product information:

```
IDF_PYTHON_ENV_PATH: C:\Espressif\python_env\idf5.3_py3.11_env

Operating System: windows 11
Java Runtime Version: 17.0.6+10-LTS
Eclipse Version: 4.32.0.v20240601-0610
Eclipse CDT Version: 11.6.1.202406241230
IDF Eclipse Plugin Version: 3.1.0.202410070705
ERROR: C:\Users\fredv\.espressif\espidf.constraints.v5.3.txt doesn't exist. Perhaps you've forgotten to run the install scripts. Please check the installation guide for more information.

ESP-IDF v5.3.1-dirty
Python set for IDF_PYTHON_ENV: Python 3.11.2
```
As suggested I looked in Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/build.cmake:365.  It says:

```
        message(STATUS "Checking Python dependencies...")
        execute_process(COMMAND "${python}" "${idf_path}/tools/idf_tools.py" "check-python-dependencies"
            RESULT_VARIABLE result)
        if(result EQUAL 1)
            # check_python_dependencies returns error code 1 on failure
            message(FATAL_ERROR "Some Python dependencies must be installed. Check above message for details.")
```

The message that I see is: 
```
C:\Users\fredv\.espressif\espidf.constraints.v5.3.txt doesn't exist. Perhaps you've forgotten to run the install scripts. Please check the installation guide for more information.
```

But I have no clue where to look. 

Some help will be appreciated. Thanks!

goofy2k, 2024-11-16:
I looked at the installation guide [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/windows-setup.html).

There it is suggested to run the ESP-IDF command prompt. When I start that, I get:
```
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\
Python 3.11.2
Using Git in C:\Espressif\tools\idf-git\2.44.0\cmd\
git version 2.44.0.windows.1
Checking Python compatibility
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v5.3.1

Adding ESP-IDF tools to PATH...
WARNING: The following issue occurred while accessing the ESP-IDF version file in the Python environment: [Errno 2] No such file or directory: 'C:\\Espressif\\python_env\\idf5.3_py3.11_env\\idf_version.txt'. (Diagnostic information. It can be ignored.)
    C:\Espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin
    C:\Espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin
    C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin
    C:\Espressif\tools\cmake\3.24.0\bin
    C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.11.1\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64
    C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
    C:\Espressif\frameworks\esp-idf-v5.3.1\tools

Checking if Python packages are up to date...
Constraint file: C:\Espressif\espidf.constraints.v5.3.txt
Requirement files:
 - C:\Espressif\frameworks\esp-idf-v5.3.1\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe
Python requirements are satisfied.

Detected installed tools that are not currently used by active ESP-IDF version.
For removing old versions of amazon-corretto-11-x64-windows-jdk, esp-clang, esp-rom-elfs, esp32ulp-elf, espressif-ide, idf-driver, idf-python-wheels, ninja, openocd-esp32, riscv32-esp-elf, riscv32-esp-elf-gdb, xtensa-esp-elf, xtensa-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf use command 'python.exe C:\Espressif\frameworks\esp-idf-v5.3.1\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe C:\Espressif\frameworks\esp-idf-v5.3.1\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
```

goofy2k, 2024-11-16:
The strange thing is that in the command promt the system says that the Python requirements are satisfied:
```
Checking if Python packages are up to date...
Constraint file: C:\Espressif\espidf.constraints.v5.3.txt
Requirement files:
 - C:\Espressif\frameworks\esp-idf-v5.3.1\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe
Python requirements are satisfied.
```

Whereas the Espressif-IDE complains that they are not!

goofy2k, 2024-11-16:
I found out that I can create a new folder and select that in the Espressif-IDE with File > Switch Workspace > Other...

Then new ESP-IDE projects can be created in there or Existing projects can be imported. Upon builiding, everything goes well!

More clear directions to solve this kind of problems would be helpful!

pawel6843, 2024-11-27:
I have something similar, but the problem occures only when I try to run the SDK Configuration Editor in Visual Studio.
ERROR: C:\Users\${USER}\.espressif\espidf.constraints.v5.3.txt doesn't exist. Perhaps you've forgotten to run the install scripts. Please check the installation guide for more information.
Generally speaking I had already other older IDF installed, but this happens with IDF 5.3.1.

pawel6843, 2024-11-27:
Ah ok it was building in VS only because it was still using repo of the old IDF. When I forced it tu use new 5.3.1, it is not building: Tool doesn't match supported version from list ['esp-13.2.0_20240530']:
  C:/IDF_52/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
fullclean does not help.
Bud building outside VS works.

pawel6843, 2024-11-27:
Ah here I found a solution for me: https://esp32.com/viewtopic.php?t=33031

goofy2k, 2025-01-12:
I progressively changed to working in a terminal window using idf.py. So using command line commands.  This works much stabler than in the Espressif-IDE.  I edit my files in Visual Studio 2022. This has a lot of searching, formatting, editing functionality and good visibility.

closed 2025-01-12 by goofy2k