espressif/idf-eclipse-plugin #1083: Cannot enter into debug mode

GayathriHommi, 2024-11-29:
**Describe the bug**
I'm using esp32h2 and Expressif IDE. I need to debug a project. But i'm facing issue to enter into debug mode

**To Reproduce**
Steps to reproduce the behavior:
1. I'm following this document to debug my esp32h2 board with buildin USB-tag.  Create a new Expressif IDF project
2. Build the project. And click on debug as>> debug configuration. Click on GDB hardware debugging. On the Right side click on debugger and change the GDB command to "riscv32-esp-elf-gdb", connection to "localhost:3333", Go to Start up and add "mon reset halt
maintenance flush register-cache
set remote hardware-watchpoint-limit 2" in initiation command. Scroll down and select set break point at : app_main
3. Then click apply and debug
4.
Error occured

5. If i change the debug server to general serial and select com port. The launch configuration is taking for ever to load

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Debug Configuration settings
[image]
[image]
[image]

**Espressif-IDE Product Information:**
[1/4] cmd.exe /C "cd /D D:\IDF_ws\Hello_world\blink\build\esp-idf\esptool_py && C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v5.3.1/components/partition_table/check_sizes.py --offset 0x8000 partition --type app D:/IDF_ws/Hello_world/blink/build/partition_table/partition-table.bin D:/IDF_ws/Hello_world/blink/build/blink.bin"
blink.bin binary size 0x32130 bytes. Smallest app partition is 0x100000 bytes. 0xcded0 bytes (80%) free.

[2/4] cmd.exe /C "cd /D D:\IDF_ws\Hello_world\blink\build\bootloader && C:\Espressif\tools\cmake\3.24.0\bin\cmake.exe --build ."
[1/1] cmd.exe /C "cd /D D:\IDF_ws\Hello_world\blink\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v5.3.1/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 D:/IDF_ws/Hello_world/blink/build/bootloader/bootloader.bin"

Bootloader binary size 0x53d0 bytes. 0x2c30 bytes (35%) free.

Build complete (0 errors, 0 warnings): D:\IDF_ws\Hello_world\blink\build
C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-v5.3.1\tools\idf_size.py D:/IDF_ws/Hello_world/blink/build/blink.map
                            Memory Type Usage Summary                             
+--------------------------------------------------------------------------------+
| Memory Type/Section | Used [bytes] | Used [%] | Remain [bytes] | Total [bytes] |
|---------------------+--------------+----------+----------------+---------------|
| Flash Code          |       154118 |     1.84 |        8234458 |       8388576 |
|    .text            |       110378 |     1.32 |                |               |
|    .rodata          |        43484 |     0.52 |                |               |
|    .appdesc         |          256 |      0.0 |                |               |
| DIRAM               |        55156 |    21.38 |         202844 |        258000 |
|    .text            |        44246 |    17.15 |                |               |
|    .data            |         6625 |     2.57 |                |               |
|    .bss             |         4272 |     1.66 |                |               |
| LP SRAM             |           24 |     0.59 |           4072 |          4096 |
|    .rtc_reserved    |           24 |     0.59 |                |               |
+--------------------------------------------------------------------------------+
C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v5.3.1/\components\partition_table\gen_esp32part.py D:\IDF_ws\Hello_world\blink\build\partition_table\partition-table.bin
Total time taken to build the project: 455 ms

**Eclipse Error log:** 

 Error in final launch sequence:

Failed to execute MI command:
-target-select remote localhost:3333

Error message from debugger back end:
localhost:3333: Connection timed out.
Failed to execute MI command:
-target-select remote localhost:3333

Error message from debugger back end:
localhost:3333: Connection timed out.
localhost:3333: Connection timed out.

kolipakakondal, 2024-11-29:
Hi @GayathriHommi Could you try **ESP-IDF OpenOCD Debugging** and see how it goes.

https://docs.espressif.com/projects/espressif-ide/en/latest/openocddebugging.html#openocddebugging

GayathriHommi, 2024-12-01:
Thank you a lot @kolipakakondal. Using OpenOCD debugging worked out.

GayathriHommi, 2024-12-02:
@kolipakakondal can I use esp-matter in esp idf for esp32h2