espressif/idf-eclipse-plugin #1087: ESP-IDF Manager - Add ESP-IDF keeps failing

lbthomsen, 2024-12-03:
Latest version of ESP-IDF - git clone and git checkout release/v5.4.  Latest version of Espressif-IDE (also tried clean Eclipse w. plugin)

ESP-IDF working, install.sh has been run, export.sh has been run, idf.py working flawlessly.  Adding in Eclipse result in failure:

Executing /home/lth/.espressif/python_env/idf5.4_py3.12_env/bin/python3 /home/lth/esp/esp-idf/tools/idf_tools.py export --format=key-value
Executing null/bin/python /home/lth/esp/esp-idf/tools/idf.py --version
Executing null/bin/python /home/lth/esp/esp-idf/tools/idf.py --list-targets

It does NOT matter which version - adding existing or downloading one - same result.  There is another closed bug describing this (#1040) - I see exactly the same.  Base OS is an up-to-date Debian.

lbthomsen, 2024-12-03:
Running the first command manually result in this:

```
lth@nb7:~/esp/esp-idf$ /home/lth/.espressif/python_env/idf5.4_py3.12_env/bin/python3 /home/lth/esp/esp-idf/tools/idf_tools.py export --format=key-value
PATH=/home/lth/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/home/lth/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:/home/lth/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin:/home/lth/.espressif/tools/esp-clang/esp-18.1.2_20240912/esp-clang/bin:/home/lth/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20240906/riscv32-esp-elf/bin:/home/lth/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/home/lth/.espressif/tools/cmake/3.30.2/bin:/home/lth/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:/home/lth/.espressif/tools/ninja/1.12.1/:/home/lth/.espressif/tools/qemu-xtensa/esp_develop_9.0.0_20240606/qemu/bin:/home/lth/.espressif/tools/qemu-riscv32/esp_develop_9.0.0_20240606/qemu/bin:$PATH
IDF_DEACTIVATE_FILE_PATH=/tmp/tmpnhdjv0kbidf_6757
```

sim0njo, 2024-12-31:
Hi, any progress or workaround on this topic?

alirana01, 2025-01-07:
Possible duplication here 
https://github.com/espressif/idf-eclipse-plugin/issues/1105
@sigmaaa

alirana01, 2025-01-22:
Closing here as duplication under this
https://github.com/espressif/idf-eclipse-plugin/issues/1105

closed 2025-01-22 by alirana01