espressif/idf-eclipse-plugin #1102: espressif-IDE editor lacks code refactor tools

emaayan, 2024-12-20:
**Describe the bug**
A clear and concise description of what the bug is.

this is the result of switching espressif switching to the new LSP code, because those refactors exists in the legacy code, 
I've already opened an issue with cdt https://github.com/eclipse-cdt/cdt-lsp/issues/386 but they partially referred me to clangd 
I'm not sure how can this be solved, but considering eclispe's massive refactoring abilities it's amazing it's being downgraded to visual studio's refactor feature just because you switched editors. 

**To Reproduce**
Steps to reproduce the behavior:
1. mark a code block 
2. RCM on it,  and highlight refactor 

**Expected behavior**
having access to code factor tools 

**Screenshots**
[image]
[image]

alirana01, 2025-01-07:
Hi @emaayan 
Thanks for bringing forward the issue as you mentioned LSP4E project lacks some integrations mentioned by the bug reported by you on cdt-lsp. We are working on such issues that are coming forward via Espressif-IDE since we started rolling out the LSP editor. One main reason to do this was to move to clangd based indexing as the CDT indexer mentioned by Jonah was not indexing things properly and was causing issues. Along with that we are trying to bridge the gap with old editor. We appreciate your patience and understanding in the matter

emaayan, 2025-01-09:
> Hi @emaayan Thanks for bringing forward the issue as you mentioned LSP4E project lacks some integrations mentioned by the bug reported by you on cdt-lsp. We are working on such issues that are [...]

i suspect this indexer maybe causing issues as well, there have been many cases where for some reason it would be "out of sync" with the code and i will get a compiler error where there should be none
even cases such commenting like 
ESP_LOGI(TAG,"HELLO)
if i do 
//ESP_LOGI(TAG,"HELLO) 
i get a compiler error 
then i go back do
// ESP_LOGI(TAG,"HELLO) 
and it's ok.