espressif/idf-eclipse-plugin #1105: Unable to install ESP-IDF 5.3.2 via ESP-IDF Manager in Elclipse

sim0njo, 2024-12-28:
**Describe the bug**
Unable to install ESP-IDF from ESP-IDF Manager, tried downloading it, tried using a downloaded version, tried it 10 times
This really annoyes me
Your documentation about installing the ESP-IDF in Eclipse is also like going in circles, missing concise and working procedure !!!

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'ESP-IDF manager' in Eclipse
2. Click on 'Add ESP-IDF' -> download ESP-IDF to C:\Espressif -> press Finish
3. Scroll down to '....'
4. See error
>>>
Selected targets are: esp32c6, esp32h2, esp32s3, esp32c61, esp32c2, esp32c3, esp32c5, esp32, esp32p4, esp32s2
Current system platform: win64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, esp-clang, riscv32-esp-elf, esp32ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util, esp-rom-elfs, qemu-xtensa, qemu-riscv32
Skipping xtensa-esp-elf-gdb@14.2_20240403 (already installed)
Skipping riscv32-esp-elf-gdb@14.2_20240403 (already installed)
Skipping xtensa-esp-elf@esp-13.2.0_20240530 (already installed)
Skipping esp-clang@16.0.1-fe4f10a809 (already installed)
Skipping riscv32-esp-elf@esp-13.2.0_20240530 (already installed)
Skipping esp32ulp-elf@2.38_20240113 (already installed)
Skipping cmake@3.30.2 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20241016 (already installed)
Skipping ninja@1.12.1 (already installed)
Skipping idf-exe@1.0.3 (already installed)
Skipping ccache@4.10.2 (already installed)
Skipping dfu-util@0.11 (already installed)
Skipping esp-rom-elfs@20240305 (already installed)
Skipping qemu-xtensa@esp_develop_9.0.0_20240606 (already installed)
Skipping qemu-riscv32@esp_develop_9.0.0_20240606 (already installed)

Executing C:\Users\simonjo\AppData\Local\Microsoft\WindowsApps\python3.EXE C:\Espressif\esp-idf-v5.3.2\tools\idf_tools.py install-python-env
Python 3.11.2
pip 24.3.1 from C:\Users\simonjo\Espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\pip (python 3.11)

Skipping the download of C:\Users\simonjo\.espressif\espidf.constraints.v5.3.txt because it was downloaded recently.
Upgrading pip and setuptools...
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Requirement already satisfied: pip in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (24.3.1)
Requirement already satisfied: setuptools in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (71.0.0)
Installing Python packages
 Constraint file: C:\Users\simonjo\.espressif\espidf.constraints.v5.3.txt
 Requirement files:
  - C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi, https://dl.espressif.com/pypi
Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
Requirement already satisfied: setuptools in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 3)) (71.0.0)
Requirement already satisfied: packaging in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 4)) (24.2)
Requirement already satisfied: click in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 7)) (8.1.8)
Requirement already satisfied: pyserial in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 8)) (3.5)
Requirement already satisfied: cryptography in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 9)) (42.0.8)
Requirement already satisfied: pyparsing in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 10)) (3.2.0)
Requirement already satisfied: pyelftools in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 11)) (0.31)
Requirement already satisfied: idf-component-manager in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 12)) (2.1.1)
Requirement already satisfied: esp-coredump in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 13)) (1.12.0)
Requirement already satisfied: esptool in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 14)) (4.8.1)
Requirement already satisfied: esp-idf-kconfig in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 15)) (2.4.1)
Requirement already satisfied: esp-idf-monitor in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 16)) (1.5.0)
Requirement already satisfied: esp-idf-nvs-partition-gen in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 17)) (0.1.6)
Requirement already satisfied: esp-idf-size in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 18)) (1.6.1)
Requirement already satisfied: esp-idf-panic-decoder in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 19)) (1.2.1)
Requirement already satisfied: pyclang in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 20)) (0.6.0)
Requirement already satisfied: construct in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 21)) (2.10.70)
Requirement already satisfied: freertos_gdb in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 24)) (1.0.3)
Requirement already satisfied: colorama in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from click->-r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 7)) (0.4.6)
Requirement already satisfied: cffi>=1.12 in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from cryptography->-r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 9)) (1.17.1)
Requirement already satisfied: pyyaml in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager->-r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 12)) (6.0.2)
Requirement already satisfied: ruamel.yaml in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager->-r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 12)) (0.18.6)
Requirement already satisfied: requests in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager->-r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 12)) (2.32.3)
Requirement already satisfied: requests-file in c:\users\simonjo\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager->-r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 12)) (2.1.0)
[... 24 lines cut ...]
[... cut ...]
**Expected behavior**
ESP-IDF to be installed. Come on guys this problem is ongoing for quiet some time now

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

sim0njo, 2024-12-31:
I try to find some logic in the failures, so tried maybe 10 times and found following:

1) First installed Eclipse CDT and ESP-IDF Plugin

2) in ESP-IDF Manager -> Add ESP-IDF
- GIT = C:\Program Files\Git\cmd\git.EXE
- Python = C:\Users\simonjo\AppData\Local\Microsoft\WindowsApps\python3.EXE
- Download v5.3.2 to c:\Espressif

3) This ends with following error and indicates a script error as there is no frameworks sub-directory !!!
>>>
ERROR: It is not possible to determine the IDF version: [Errno 2] No such file or directory: 'C:\\Espressif\\frameworks\\esp-idf-v5.3.2\\components\\esp_common\\include\\esp_idf_version.h'
<<<

4) I create c:\Espressif\frameworks and copy the esp-idf-v5.3.2 directory to it (copy, not move)

5) Redo Add ESP-IDF from ESP-IDF Manager
- Use existing ESP-IDF directory -> c:\Espressif\esp-idf-v5.3.2
- Now the different tools are installed
- But it still ends with an error like this
>>>

Executing C:\Users\simonjo\AppData\Local\Microsoft\WindowsApps\python3.EXE C:\Espressif\esp-idf-v5.3.2\tools\idf_tools.py export --format=key-value
Executing null/Scripts/python.exe C:\Espressif\esp-idf-v5.3.2\tools\idf.py --version
Executing null/Scripts/python.exe C:\Espressif\esp-idf-v5.3.2\tools\idf.py --list-targets

Cannot run program "null/Scripts/python.exe" (in directory "\"): CreateProcess error=2, The system cannot find the file specified
<<<

Again a scripting problem, on the first line you see python3 being called from the corect path. But in line 2 and 3 there is a problem. This should refer to c:\Users\simonjo\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe

I tried to create c:\null and copy the Scripts directory there... but Add ESP-IDF keeps failing !

sim0njo, 2024-12-31:
Looking at idf_tools.py I find following code

def get_python_exe_and_subdir() -> Tuple[str, str]:
    """
    Returns:
    * python executable name (python on Unix, python.exe on Win)
    * subdir in which the executable is (bin for Unix, Scripts for Win)
    """
    if sys.platform == 'win32':
        subdir = 'Scripts'
        python_exe = 'python.exe'
    else:
        subdir = 'bin'
        python_exe = 'python'
    return python_exe, subdir

def get_python_env_path() -> Tuple[str, str, str, str]:
    """
    Returns tuple of Python environment path, Python env. path with subdir and full path from Python (i.e. with executable).
    """
    idf_version = get_idf_version()
    idf_python_env_path = os.getenv('IDF_PYTHON_ENV_PATH') or os.path.join(g.idf_tools_path,
                                                                           'python_env',
                                                                           PYTHON_VENV_DIR_TEMPLATE.format(idf_version,
                                                                                                           PYTHON_VER_MAJOR_MINOR))

    python_exe, subdir = get_python_exe_and_subdir()
    idf_python_export_path = os.path.join(idf_python_env_path, subdir)
    virtualenv_python = os.path.join(idf_python_export_path, python_exe)

    return idf_python_env_path, idf_python_export_path, virtualenv_python, idf_version

So it all comes down to IDF_PYTHON_ENV_PATH... which is set to following: IDF_PYTHON_ENV_PATH=C:\Users\simonjo\Espressif\python_env\idf5.3_py3.11_env

From my calculation this should end up in
virtualenv_python=c:\Users\simonjo\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe

sigmaaa, 2025-01-06:
Hi @sim0njo,

would it be possible to connect on MS Teams to work through this issue together? The problem is quite unusual, and we have not been able to reproduce it on our end. If this works for you, please write to [ide.support@espressif.com](mailto:ide.support@espressif.com) with your time zone and a suitable time. We will then send you an invite. Thank you

sim0njo, 2025-01-07:
I deleted the IDF_PYTHON_ENV_PATH from windows env variables and retried setup of ESP-IDF 5.3.2 with following params:

Git: C:\Program Files\Git\cmd\git.EXE
Python: C:\Users\simonjo\AppData\Local\Microsoft\WindowsApps\python3.EXE
Download: v5.3.2
Download dir: C:\Users\simonjo\esp-temp
Press "Finish" button 

Now I get below response:

ERROR: It is not possible to determine the IDF version: [Errno 2] No such file or directory: 'C:\\Espressif\\frameworks\\esp-idf-v5.3.2\\components\\esp_common\\include\\esp_idf_version.h'

Executing C:\Users\simonjo\AppData\Local\Microsoft\WindowsApps\python3.EXE C:\Users\simonjo\esp-temp\esp-idf-v5.3.2\tools\idf_tools.py install-python-env
ERROR: It is not possible to determine the IDF version: [Errno 2] No such file or directory: 'C:\\Espressif\\frameworks\\esp-idf-v5.3.2\\components\\esp_common\\include\\esp_idf_version.h'

Executing C:\Users\simonjo\AppData\Local\Microsoft\WindowsApps\python3.EXE C:\Users\simonjo\esp-temp\esp-idf-v5.3.2\tools\idf_tools.py export --format=key-value
Traceback (most recent call last):
  File "C:\Users\simonjo\esp-temp\esp-idf-v5.3.2\tools\idf_tools.py", line 3239, in <module>
    main(sys.argv[1:])
  File "C:\Users\simonjo\esp-temp\esp-idf-v5.3.2\tools\idf_tools.py", line 3231, in main
    action_func(args)
  File "C:\Users\simonjo\esp-temp\esp-idf-v5.3.2\tools\idf_tools.py", line 2154, in action_export
    tools_info = load_tools_info()
  File "C:\Users\simonjo\esp-temp\esp-idf-v5.3.2\tools\idf_tools.py", line 1607, in load_tools_info
    with open(tool_versions_file_name, 'r') as f:  # type: ignore
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Espressif\\frameworks\\esp-idf-v5.3.2\\tools/tools.json'

sim0njo, 2025-01-11:
In the meantime I have installed Visual Studio Code with the ESP-IDF Plugin van v5.3.2... and it is working just fine !

So I will stick to this solution as I have time restrictions to move to my new PC platform and I already lost 2 weeks with Eclipse.

vortex-surfer, 2025-02-03:
I am also running into this error with Espressif IDE 3.2.0 on Linux (Ubuntu 24.04) trying to install either ESP-IDF v5.3.1, v5.3.2 or v5.4. Setting/unsetting system environment variables does not change any behaviour it always leads to the same error:
```
Executing /home/wome/.espressif/python_env/idf5.3_py3.12_env/bin/python3 /opt/dev/espressif/frameworks/esp-idf-v5.3.1/tools/idf_tools.py export --format=key-value
Executing null/bin/python /opt/dev/espressif/frameworks/esp-idf-v5.3.1/tools/idf.py --version
Executing null/bin/python /opt/dev/espressif/frameworks/esp-idf-v5.3.1/tools/idf.py --list-targets
```

flodmotorgrodan, 2025-02-18:
Me and ChatGPT has been working on the problem for a few days.

The problem is, the ESP-IFD manager installer relies on _idf_tools.py_ to return a list of environment variables generated by executing _idf_tools.py export --format=key-value_

The installer, successful until this point, now expects a path to the python environment IDF_PYTHON_ENV_PATH=C:\Espressif\python_env\idf5.3_y3.11_env, to be in the list provided by _idf_tools.py_

Here is the function that assumes a path is supplied by idf_tools.py: [ToolsJob.java#L263](https://github.com/espressif/idf-eclipse-plugin/blob/master/bundles/com.espressif.idf.ui/src/com/espressif/idf/ui/tools/ToolsJob.java#L263)

The _idf_tools.py export --format=key-value_ is a bit complex, but it looks like the [IDF_PYTHON_ENV_PATH] is only returned if a python virtual setup is detected AND the path to that python differs from the one in the operating system environment variable. If one exists. In my case, as I have IDF_PYTHON_ENV_PATH in my os environment, the variable was not returned.

Here is the logic selecting for output of [IDF_PYTHON_ENV_PATH]
[df_tools.py#L2189](https://github.com/espressif/esp-idf/blob/1160a86ba0b87b0ea68ad6b91eb10fe9c34ad692/tools/idf_tools.py#L2189)

The problem is a proper setup of ESP-IDF  [IDF_PYTHON_ENV_PATH] will generate the variables in the operating system.

So a workaround could be to to not have the python virtual specific to ESP-IDE activated and hope the installer will work using a default python system. But it still needs the ESP-IDE \frameworks\ folder to point to and the installer has the same requirements as when setting up the environment for compiling projects in Powershell.

As I do not understand the hyper complexity i just added a line 

export_vars['IDF_PYTHON_ENV_PATH'] = os.getenv('IDF_PYTHON_ENV_PATH')

closed 2025-02-28 by alirana01