espressif/idf-eclipse-plugin #1109: "Initializing language server" blocked on 28%

RicSanLom, 2025-01-08:
Hi,
I've been  using Espressif-IDE 3.1.0 20241007-0706 on WIndows 10 for some months. Starting from 7th January the progress of "Initializing language server "C/C++ Language Server" bloks on 28% and indeed the language server doesn't work.
What can I do?
Thanks for your help.
Best regards
Riccardo

sigmaaa, 2025-01-08:
Hi @RicSanLom,

The fix for this issue is available on the master branch and will be included in the next release. In the meantime, you can use a workaround: enclose the path in the `Additional` section on the clangd preference page with quotation marks, as shown below:
[image]
Or you can download artifacts with a master build below on this page: https://github.com/espressif/idf-eclipse-plugin/actions/runs/12425630976

RicSanLom, 2025-01-08:
Hi sigmaaa, quotation marks solved the problem as you wrote. Thank you very much.

sigmaaa, 2025-01-11:
fixed in[ 3.2.0 release](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.2.0)

closed 2025-01-11 by sigmaaa