espressif/idf-eclipse-plugin #1115: Espressif-IDE is not clearing serial port before trying to upload

jessecheu, 2025-01-12:
**Describe the bug**
A clear and concise description of what the bug is.

When trying to upload any program to a ESP32 MCU. I get the error message: 

`â†�[0;33m--- esp-idf-monitor 1.5.0 on \\.\COM4 115200â†�[0m
--- Warning: ELF file 'C:/Users/../build/udemy_esp32_iot_app.elf' does not exist
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
could not open port '\\\\.\\COM4': PermissionError(13, 'Access is denied.', None, 5)
--- Connection to \\.\COM4 failed. Available ports:
COM4`

I tried it on different ESP32 target devices so it seems the problem is with uploading using the espressif-IDE. 

**To Reproduce**
Steps to reproduce the behavior:
1. Trying to upload code.

**Expected behavior**
A clear and concise description of what you expected to happen.
Code is able to successfully upload code.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)
[CDT Build environment variables.txt](https://github.com/user-attachments/files/18387750/CDT.Build.environment.variables.txt)

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)

Please attach the error log as described here [!SESSION 2025-01-09 205241.515 ----.txt](https://github.com/user-attachments/files/18387749/SESSION.2025-01-09.205241.515.----.txt)

I found that the fix is going to task manager and ending all python process related to espressif-ide.
It seems that the file that isn't being closed properly before trying to upload the code to the target is.
"C:\Espressif\python_env\idf5.3_py3.11_env\Scripts" and "C:\Espressif\tools\idf-python\3.11.2"
I believe that pySerial isn't being closed properly. 
Another fix was to restart my computer which properly closed the serial port.

DerValdemar, 2025-01-21:
I have same problem. Did you found any solutions for this?

sigmaaa, 2025-01-21:
Hi @jessecheu, @DerValdemar 

Thank you for reporting this issue!

I’m unable to reproduce it on my end, but it seems the issue might occur because the serial monitor opens automatically after flashing.

When this happens, could you try switching to the terminal tab and disconnecting the terminal connection using this button?
[image]

Additionally, if opening the serial monitor after flashing is not essential, you can disable this behavior by editing the configuration and turning off the "Open Serial Monitor After Flashing" option.

[image]

However, even without this option, the serial monitor can be launched manually by using this icon:

[image]
and then selecting "ESP-IDF Serial Monitor"

PeDaLa42, 2025-02-11:
Some weeks ago, my Espressif IDE version 3.0.0 installation worked as expected:
Pressing 'Run' button in the IDE closed the COM port in the terminal view (i.e., in ESP-IDF Serial Monitor aka esp-idf-monitor), uploaded the firmware via UART (esptool.py) and re-opend the COM port in the Terminal view.

However, this doesn't work anymore and I have exactly the same issue as described by jessecheu:
1. The terminal view (esp-idf-monitor) reports "could not open port '\\\\.\\COM5': PermissionError(13, 'Zugriff verweigert', None, 5)" and 
2. esptool.py reports "A fatal error occurred: Could not open COM5, the port is busy or doesn't exist. (could not open port 'COM5': PermissionError(13, 'Zugriff verweigert', None, 5))"

My solution is not restarting the computer.
Logging off and on is enough to completely close the somehow still opened COM port.
After this, i.e., logging off and on, firmware upload with Espressif IDE works again (with NOT opened terminal view)

I investigated the issue further and my findings are (with Espressif IDE version 3.0.0):
1. The same result (non-closed COM port) happens with the following steps:
a) Start Espressif IDE.
b) In Espressif IDE: open terminal view.
c) In the terminal view: Connect with ESP-IDF Serial Monitor to COM5 (this the port where my ESP32S3 is connected to).
This opens COM5 successful and prints out some lines from my ESP32.
d) Press 'Disconnect Terminal Connection' button (N-shaped button) to close COM5 in the terminal view. 
With this, '<Closed>' appears in the COM5 window tab.
e) Open COM5 with another terminal emulation, e.g., Tera Term.
With this, at least Tera Term reports "Cannot open COM5. Access denied". 
I.e., COM5 is still opened, although ESP-IDF Serial Monitor should have closed it (see step d).
By logging off and on, the steps a) to e) can be re-done with the same result ;-)

2. The same result (non-closed COM port) happens with Espressif IDE version 3.1.0.

3. Modify step e) by NOT opening COM5 in another terminal but re-open COM5 in the terminal view of Espressif IDE (press 'Open a Terminal' button in the terminal view).
In the terminal view, ESP-IDF Serial Monitor aka esp-idf-monitor reports "could not open port '\\\\.\\COM5': PermissionError(13, 'Zugriff verweigert', None, 5)".

4. Modify step d) by NOT pressing 'Disconnect Terminal Connection' button but by pressing on the keyboard 'CTRL+T' and 'CTRL+E'. 
This closes the ESP-IDF Serial Monitor. With this, COM5 is completely closed and can be re-opened by any terminal program, e.g., Tera Term or ESP-IDF Serial Monitor aka esp-idf-monitor.

Before updating to Windows 11 version 24H2, my Espressif IDE version 3.0.0 worked as expected (see my first lines above).
So, I believe this could be a version 24H2 issue.
And, it seams that this issue has something to do with controlling esp-idf-monitor in eclipse/Espressif IDE (see my findings 4): 
- GUI button 'Disconnect Terminal Connection' in the terminal view does NOT work (does NOT close COM port).
- Keyboard buttons 'CTRL+T' and 'CTRL+E' do work (close COM port).

As I believe that controlling esp-idf-monitor in eclipse/Espressif IDE is broken, uploading firmware in combination with a running esp-idf-monitor is also broken (as described by the initial comment).

sigmaaa, 2025-03-03:
Hi @jessecheu, @DerValdemar , @PeDaLa42,

Could you please try Espressif-IDE from [this link](https://github.com/espressif/idf-eclipse-plugin/actions/runs/13628179487?pr=1168) and check if you encounter the same issue on this PR? Thank you

PeDaLa42, 2025-03-04:
I installed espressif IDE 3.3.0 and I still get the following error in the serial monitor when uploading a new firmware (i.e., pressing 'Run' button) WHILE the terminal view/ESP-IDF Serial Monitor is open.

â†�[0;33m--- Warning: GDB cannot open serial ports accessed as COMxâ†�[0m
â†�[0;33m--- Using \\.\COM5 instead...â†�[0m
â†�[0;33m--- esp-idf-monitor 1.5.0 on \\.\COM5 115200â†�[0m
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
could not open port '\\\\.\\COM5': PermissionError(13, 'Zugriff verweigert', None, 5)
--- Connection to \\.\COM5 failed. Available ports:
COM5

I.e., unfortunately, the issue is not resolved.

My work-around still works: BEFORE uploading: close the serial monitor via KEYBOARD (NOT by GUI button!)

ProgettiPascon, 2025-03-22:
Hi all. I just wanted to ask if someone has found a simpler workaround or a solution to this issue. Cause it's been driving me mad for months right now. And this is the only thread I found which speaks about it.
Even if I'd rather not do that, I might be forced to switch to VSCode with IDF extension.
Moreover I can confirm that back on WIN10, the fw upload was working straightforward, without the need of closing the terminal.

sigmaaa, 2025-03-25:
Hi @jessecheu,

Could you please check if you encounter this issue with Espressif-IDE using the following link: https://github.com/espressif/idf-eclipse-plugin/actions/runs/13628179487/artifacts/2680526448? I’ve replaced process.destroy() with process.destroyForcibly(), and I hope this resolves the issue. However, this problem is tricky, as I haven’t been able to reproduce it, so your help in testing would be greatly appreciated.

Also, as a workaround, you can terminate the Python process responsible for the serial monitor in the Task Manager.

sigmaaa, 2025-03-28:
Hi @ProgettiPascon,

Thank you for verifying it! I've made a few additional changes to the PR. Could you please check on this version https://github.com/espressif/idf-eclipse-plugin/actions/runs/14134877905/artifacts/2840886033?

ProgettiPascon, 2025-03-29:
Hi @sigmaaa, as always thank you for your help. I'm testing the latest version you provided and It's WORKING.
If anyone else dealing with this issue wants to try for himself, here's what I did:

1. Downloaded and extracted the software from the [latest version @sigmaaa provided](https://github.com/espressif/idf-eclipse-plugin/issues/1115#issuecomment-2762146544).
2. Tried to flash a test project using the same workspace folder of the previous versions I used, but it showed the same issue: the first flashing went flawlessly, the second one, having the serial terminal open, halted.
3. I've then deleted the old workspace and let the IDE create a new one.
4. Setup the ESP-IDF from the IDF-Manager menu.
5. Created a new project and flashed it 2 times as I did in step 2 and IT WORKED FLAWLESSLY! I've then done more tests just to make sure it wasn't a random thing and worked as a charm.

Moral: Install this new version and create a new workspace.

Indastri, 2025-04-28:
Hi @ProgettiPascon, if your issue was solved can you please close this issue? Thanks in advance!

closed 2025-06-06 by sigmaaa