espressif/idf-eclipse-plugin #1119: Changing CMakeLists.txt requires Eclipse to be rebooted

leenowell, 2025-01-15:
**Describe the bug**
If you change the CMakeLists.txt file then a "problem occurs" box appears and you can't build the project until you restart Eclipse

**To Reproduce**
Steps to reproduce the behavior:
1. Edit the CMakeLists.txt file of a project
2. Build the project
**Expected behavior**
A clear and concise description of what you expected to happen.
Should be able to build the project without the error
**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)
Building in: /media/Documents/Lee/Dev/NewESP32Workspace/HomeIoT/build
Configuring in: /media/Documents/Lee/Dev/NewESP32Workspace/HomeIoT/build

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log 
[CmakelistsBug.log](https://github.com/user-attachments/files/18428203/CmakelistsBug.log)

AndriiFilippov, 2025-01-16:
@leenowell  hi ! 

Could you please provide me with the following information:

- The command you tried to add to the CMakeLists.txt file.
- The version of Eclipse you are using.

I will take a look

leenowell, 2025-01-16:
Hi @AndriiFilippov 

Thanks for getting back to me.  The command doesn't matter you just need to any change (e.g. add a space) to make a change to CMakeLists.txt and it happens. 

This is the Eclipse version meeting

Eclipse IDE for C/C++ Developers (includes Incubating components)

Version: 2024-12 (4.34.0)
Build id: 20241128-0757

Thanks

Lee.

AndriiFilippov, 2025-01-16:
@leenowell 

My env:
Linux Ubuntu 22.04

I have tested this scenarios: 
 Eclipse 24-12 + Esp Plugin 3.1.0 + ESP-IDF 5.3 
 Eclipse24-12 + Esp Plugin 3.2.0 + ESP-IDF: 5.4 / 5.3 
 Eclipse24-12 + Esp Plugin 3.2.0 + ESP-IDF: 5.4 / 5.3 
I was able to build and run this cmake command: 
message(STATUS "+++++++++++++++++++++++++++++++++++")
 
Could you please provide **Product Info** logs.  go to "Espressif" menu -> Product Information -> copy console content -> and just paste it here. I want to verify if your tools are installed correctly.

leenowell, 2025-01-16:
Hi @AndriiFilippov 

I am also on Ubuntu but version 24.04.1

As requested here are the product logs

Executing /usr/bin/python3 /home/lee/esp/esp-idf/tools/idf_tools.py list
* xtensa-esp-elf-gdb: GDB for Xtensa
  - 15.2_20241112 (recommended, installed)
* riscv32-esp-elf-gdb: GDB for RISC-V
  - 15.2_20241112 (recommended, installed)
* xtensa-esp-elf: Toolchain for 32-bit Xtensa based on GCC
  - esp-14.2.0_20241119 (recommended, installed)
* esp-clang: Toolchain for all Espressif chips based on clang (optional)
  - esp-18.1.2_20240912 (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-14.2.0_20241119 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.38_20240113 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.30.2 (recommended, installed)
  - 3.16.3 (supported)
* openocd-esp32: OpenOCD for ESP32
  - v0.12.0-esp32-20241016 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.12.1 (recommended, installed)
* esp-rom-elfs: ESP ROM ELFs
  - 20241011 (recommended, installed)
* qemu-xtensa: QEMU for Xtensa (optional)
  - esp_develop_9.0.0_20240606 (recommended)
* qemu-riscv32: QEMU for RISC-V (optional)
  - esp_develop_9.0.0_20240606 (recommended)
OK

CDT Build environment variables
PATH: /home/lee/.espressif/tools/xtensa-esp-elf-gdb/15.2_20241112/xtensa-esp-elf-gdb/bin:/home/lee/.espressif/tools/riscv32-esp-elf-gdb/15.2_20241112/riscv32-esp-elf-gdb/bin:/home/lee/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin:/home/lee/.espressif/tools/esp-clang/esp-18.1.2_20240912/esp-clang/bin:/home/lee/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin:/home/lee/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/home/lee/.espressif/tools/cmake/3.30.2/bin:/home/lee/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:/home/lee/.espressif/tools/ninja/1.12.1/:/home/lee/.espressif/python_env/idf5.5_py3.12_env/bin:/home/lee/esp/esp-idf/tools:/home/lee/.local/bin:/opt/maven/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
XAUTHORITY: /run/user/1000/.mutter-Xwaylandauth.7OHB02
INVOCATION_ID: 8aed3ee9a389435fb520d2b66d5d621b
XMODIFIERS: @im=ibus
XDG_DATA_DIRS: /usr/share/ubuntu:/usr/share/gnome:/home/lee/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
GDMSESSION: ubuntu
DBUS_SESSION_BUS_ADDRESS: unix:path=/run/user/1000/bus
ESP_IDF_VERSION: 5.5
XDG_CURRENT_DESKTOP: ubuntu:GNOME
OPENOCD_SCRIPTS: /home/lee/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts
JOURNAL_STREAM: 8:26144
IDF_DEACTIVATE_FILE_PATH: /tmp/tmp_ldmdm00idf_18693
MAVEN_HOME: /opt/maven
GSM_SKIP_SSH_AGENT_WORKAROUND: true
USERNAME: lee
SESSION_MANAGER: local/leelaptop:@/tmp/.ICE-unix/3261,unix/leelaptop:/tmp/.ICE-unix/3261
LOGNAME: lee
MANAGERPID: 2959
PWD: /home/lee
IM_CONFIG_PHASE: 1
LANGUAGE: en_GB:en
GJS_DEBUG_TOPICS: JS ERROR;JS LOG
SHELL: /bin/bash
GIO_LAUNCHED_DESKTOP_FILE: /home/lee/.local/share/applications/epp.package.cpp.desktop
GNOME_DESKTOP_SESSION_ID: this-is-deprecated
IDF_COMPONENT_MANAGER: 1
GTK_MODULES: gail:atk-bridge
DEBUGINFOD_URLS: https://debuginfod.ubuntu.com 
M2_HOME: /opt/maven
PYTHON_EXE_PATH: /usr/bin/python3
SYSTEMD_EXEC_PID: 3312
XDG_SESSION_DESKTOP: ubuntu
GNOME_SETUP_DISPLAY: :1
SHLVL: 0
MEMORY_PRESSURE_WRITE: c29tZSAyMDAwMDAgMjAwMDAwMAA=
OXYGEN_DISABLE_INNER_SHADOWS_HACK: 1
QT_IM_MODULE: ibus
JAVA_HOME: /usr/lib/jvm/default-java
MEMORY_PRESSURE_WATCH: /sys/fs/cgroup/user.slice/user-1000.slice/user@1000.service/session.slice/org.gnome.Shell@wayland.service/memory.pressure
XDG_CONFIG_DIRS: /etc/xdg/xdg-ubuntu:/etc/xdg
LANG: en_GB.UTF-8
XDG_SESSION_TYPE: wayland
IDF_TOOLS_PATH: /home/lee/.espressif
DISPLAY: :0
WAYLAND_DISPLAY: wayland-0
XDG_SESSION_CLASS: user
_: /usr/bin/gnome-session
QTWEBENGINE_DICTIONARIES_PATH: /usr/share/hunspell-bdic/
LIBOVERLAY_SCROLLBAR: 0
IDF_PATH: /home/lee/esp/esp-idf
DESKTOP_SESSION: ubuntu
GDK_CORE_DEVICE_EVENTS: 1
USER: lee
XDG_MENU_PREFIX: gnome-
GIO_LAUNCHED_DESKTOP_FILE_PID: 87624
QT_ACCESSIBILITY: 1
GJS_DEBUG_OUTPUT: stderr
SSH_AUTH_SOCK: /run/user/1000/keyring/ssh
IDF_MAINTAINER: 1
GNOME_SHELL_SESSION_MODE: ubuntu
XDG_RUNTIME_DIR: /run/user/1000
IDF_PYTHON_ENV_PATH: /home/lee/.espressif/python_env/idf5.5_py3.12_env
HOME: /home/lee
ESP_ROM_ELF_DIR: /home/lee/.espressif/tools/esp-rom-elfs/20241011/

Operating System: linux
Java Runtime Version: 21.0.5+11-Ubuntu-1ubuntu124.04
Eclipse Version: 4.34.0.v20241120-1800
Eclipse CDT Version: 11.6.1.202406241230
IDF Eclipse Plugin Version: 3.2.0.202412171009
ESP-IDF v5.5-dev-1050-gb5ac4fbdf9
Python set for IDF_PYTHON_ENV: Python 3.12.3

AndriiFilippov, 2025-01-17:
@leenowell hi ! 

I did replicate your environment. Here are my settings:
```
* xtensa-esp-elf-gdb: GDB for Xtensa
  - 15.2_20241112 (recommended, installed)
* riscv32-esp-elf-gdb: GDB for RISC-V
  - 15.2_20241112 (recommended, installed)
* xtensa-esp-elf: Toolchain for 32-bit Xtensa based on GCC
  - esp-14.2.0_20241119 (recommended, installed)
* esp-clang: Toolchain for all Espressif chips based on clang (optional)
  - esp-18.1.2_20240912 (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-14.2.0_20241119 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.38_20240113 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.30.2 (recommended, installed)
  - 3.16.3 (supported)
* openocd-esp32: OpenOCD for ESP32
  - v0.12.0-esp32-20241016 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.12.1 (recommended, installed)
* esp-rom-elfs: ESP ROM ELFs
  - 20241011 (recommended, installed)
* qemu-xtensa: QEMU for Xtensa (optional)
  - esp_develop_9.0.0_20240606 (recommended)
* qemu-riscv32: QEMU for RISC-V (optional)
  - esp_develop_9.0.0_20240606 (recommended)
OK
 
CDT Build environment variables
PATH: /home/andrii/.espressif/tools/xtensa-esp-elf-gdb/15.2_20241112/xtensa-esp-elf-gdb/bin:/home/andrii/.espressif/tools/riscv32-esp-elf-gdb/15.2_20241112/riscv32-esp-elf-gdb/bin:/home/andrii/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin:/home/andrii/.espressif/tools/esp-clang/esp-18.1.2_20240912/esp-clang/bin:/home/andrii/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin:/home/andrii/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/home/andrii/.espressif/tools/cmake/3.30.2/bin:/home/andrii/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:/home/andrii/.espressif/tools/ninja/1.12.1/:/home/andrii/.espressif/python_env/idf5.5_py3.12_env/bin:/home/andrii/Documents/ED/esp-idf-master/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
INVOCATION_ID: 31d06b823115460fa4f1854958ab4f80
XAUTHORITY: /run/user/1000/.mutter-Xwaylandauth.67WD02
XMODIFIERS: @im=ibus
XDG_DATA_DIRS: /usr/share/ubuntu:/usr/share/gnome:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
GDMSESSION: ubuntu
DBUS_SESSION_BUS_ADDRESS: unix:path=/run/user/1000/bus,guid=2101c67d29858f4e5666b1ba678a5d32
ESP_IDF_VERSION: 5.5
XDG_CURRENT_DESKTOP: ubuntu:GNOME
OPENOCD_SCRIPTS: /home/andrii/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts
JOURNAL_STREAM: 8:19446
IDF_DEACTIVATE_FILE_PATH: /tmp/tmpwr2kggm6idf_31935
GSM_SKIP_SSH_AGENT_WORKAROUND: true
USERNAME: andrii
SESSION_MANAGER: local/andrii-ThinkPad-T14-Gen-1:@/tmp/.ICE-unix/3038,unix/andrii-ThinkPad-T14-Gen-1:/tmp/.ICE-unix/3038
LOGNAME: andrii
MANAGERPID: 2689
PWD: /home/andrii
IM_CONFIG_PHASE: 1
SHELL: /bin/bash
GNOME_DESKTOP_SESSION_ID: this-is-deprecated
IDF_COMPONENT_MANAGER: 1
GTK_MODULES: gail:atk-bridge
DEBUGINFOD_URLS: https://debuginfod.ubuntu.com/
PYTHON_EXE_PATH: /usr/bin/python3
DBUS_STARTER_ADDRESS: unix:path=/run/user/1000/bus,guid=2101c67d29858f4e5666b1ba678a5d32
SYSTEMD_EXEC_PID: 3038
XDG_SESSION_DESKTOP: ubuntu
GNOME_SETUP_DISPLAY: :1
SHLVL: 0
MEMORY_PRESSURE_WRITE: c29tZSAyMDAwMDAgMjAwMDAwMAA=
OXYGEN_DISABLE_INNER_SHADOWS_HACK: 1
QT_IM_MODULE: ibus
MEMORY_PRESSURE_WATCH: /sys/fs/cgroup/user.slice/user-1000.slice/user@1000.service/app.slice/app-gnome\x2dsession\x2dmanager.slice/gnome-session-manager@ubuntu.service/memory.pressure
XDG_CONFIG_DIRS: /etc/xdg/xdg-ubuntu:/etc/xdg
LANG: en_US.UTF-8
XDG_SESSION_TYPE: wayland
IDF_TOOLS_PATH: /home/andrii/.espressif
DISPLAY: :0
WAYLAND_DISPLAY: wayland-0
XDG_SESSION_CLASS: user
_: /usr/bin/gnome-session
LIBOVERLAY_SCROLLBAR: 0
IDF_PATH: /home/andrii/Documents/ED/esp-idf-master
DESKTOP_SESSION: ubuntu
GDK_CORE_DEVICE_EVENTS: 1
USER: andrii
XDG_MENU_PREFIX: gnome-
GIO_LAUNCHED_DESKTOP_FILE_PID: 29292
QT_ACCESSIBILITY: 1
SSH_AUTH_SOCK: /run/user/1000/keyring/ssh
IDF_MAINTAINER: 1
GNOME_SHELL_SESSION_MODE: ubuntu
XDG_RUNTIME_DIR: /run/user/1000
IDF_PYTHON_ENV_PATH: /home/andrii/.espressif/python_env/idf5.5_py3.12_env
HOME: /home/andrii
DBUS_STARTER_BUS_TYPE: session
ESP_ROM_ELF_DIR: /home/andrii/.espressif/tools/esp-rom-elfs/20241011/
 
Operating System: linux
Java Runtime Version: 21.0.5+11-LTS
Eclipse Version: 4.34.0.v20241120-1800
Eclipse CDT Version: 11.6.1.202406241230
IDF Eclipse Plugin Version: 3.2.0.202501110717
ESP-IDF v5.5-dev-1428-g0f0068fff3
Python set for IDF_PYTHON_ENV: Python 3.12.3
```

[image]

I couldn't reproduce your issue. I recorded a video for you showing how I tested it. 
I based it on the standard `hello_world` example from ESP-IDF.

https://github.com/user-attachments/assets/ba5b1dc3-742d-42d0-9dac-06ff33221177

I see you're encountering the issue in a custom project. Please try testing the problem using the standard examples from ESP-IDF and let me know the results.

leenowell, 2025-01-17:
Hi @AndriiFilippov 

The video helped thanks.  I did exactly the same as you by creating a new hello_world, letting it compile and then edited the CMakeFileLists.txt file at the same level as main.  I then added a space to the end of the following line

cmake_minimum_required(VERSION 3.16) 

I then pressed the build button and reproduced the error.  I took a video of it but it was too big (100Mb+).

leenowell, 2025-01-17:
@AndriiFilippov I managed to reduce the size of the video so hopefully you can see it ok

https://github.com/user-attachments/assets/483bdcc3-8ef5-45fe-a126-c5cc35f659bb

leenowell, 2025-01-21:
Hi @AndriiFilippov 

Any update on this please?

AndriiFilippov, 2025-01-21:
@leenowell hi !

I was not able to reproduce your issue. Please, send me error logs. 
Create new project - perform your test and then export error logs. 
Brief tutorial, where to find error logs:

https://github.com/user-attachments/assets/b50201c2-c738-4cd7-918c-a5b0489b35a9

leenowell, 2025-01-21:
HI @AndriiFilippov 

As requested please find attached the error log

[EclipseESPError_log.txt](https://github.com/user-attachments/files/18494277/EclipseESPError_log.txt)

AndriiFilippov, 2025-01-21:
@leenowell  

execute this command from Terminal and then retry modify CMakeList.txt:

sudo chmod -R 755 sudo chmod -R 755 /media/Documents/Lee/Dev/NewESP32Workspace/

leenowell, 2025-01-21:
@AndriiFilippov 

It is a little more tricky than that as this file system is on an NFS share.  If you are thinking this might be a file permission issue, I have checked all the files / directories in the structure and they are all 777

sigmaaa, 2025-01-21:
Hi @leenowell,

Could you please try downloading the Espressif-IDE from this link: https://github.com/espressif/idf-eclipse-plugin/actions/runs/12691399517 and check if the issue persists?

leenowell, 2025-01-21:
Hi @sigmaaa 

I tried installing it but am struggling to get it to work.  I am on Ubuntu 24.04 and downloaded [Espressif-IDE-3.2.0-linux.gtk.x86_64] and extracted it into a directory.  I run espressif_ide and using ESP_IDF manager tried to install the idf by pointing it to my existing espidf directory and using the same location for git and python as my eclipse setup ( ).  It starts to set up the environment and I get the following error window

[image]

The log in the console is

`
Installing tools...
This can take a while. Please be patient.
Executing /usr/bin/python3 /home/lee/esp/esp-idf/tools/idf_tools.py install all
Updating /home/lee/.espressif/idf-env.json
Selected targets are: esp32c6, esp32h21, esp32c3, esp32c61, esp32c2, esp32c5, esp32h2, esp32s3, esp32s2, esp32, esp32p4
Current system platform: linux-amd64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, esp-clang, riscv32-esp-elf, esp32ulp-elf, cmake, openocd-esp32, ninja, esp-rom-elfs, qemu-xtensa, qemu-riscv32
Skipping xtensa-esp-elf-gdb@15.2_20241112 (already installed)
Skipping riscv32-esp-elf-gdb@15.2_20241112 (already installed)
Skipping xtensa-esp-elf@esp-14.2.0_20241119 (already installed)
Skipping esp-clang@esp-18.1.2_20240912 (already installed)
Skipping riscv32-esp-elf@esp-14.2.0_20241119 (already installed)
Skipping esp32ulp-elf@2.38_20240113 (already installed)
Skipping cmake@3.30.2 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20241016 (already installed)
Skipping ninja@1.12.1 (already installed)
Skipping esp-rom-elfs@20241011 (already installed)
Installing qemu-xtensa@esp_develop_9.0.0_20240606
Using GitHub assets mirror for URL: https://github.com/espressif/qemu/releases/download/esp-develop-9.0.0-20240606/qemu-xtensa-softmmu-esp_develop_9.0.0_20240606-x86_64-linux-gnu.tar.xz => https://dl.espressif.com/github_assets/espressif/qemu/releases/download/esp-develop-9.0.0-20240606/qemu-xtensa-softmmu-esp_develop_9.0.0_20240606-x86_64-linux-gnu.tar.xz
file qemu-xtensa-softmmu-esp_develop_9.0.0_20240606-x86_64-linux-gnu.tar.xz is already downloaded
Extracting /home/lee/.espressif/dist/qemu-xtensa-softmmu-esp_develop_9.0.0_20240606-x86_64-linux-gnu.tar.xz to /home/lee/.espressif/tools/qemu-xtensa/esp_develop_9.0.0_20240606
ERROR: tool qemu-xtensa version esp_develop_9.0.0_20240606 is installed, but getting error: non-zero exit code (127) with message: /home/lee/.espressif/tools/qemu-xtensa/esp_develop_9.0.0_20240606/qemu/bin/qemu-system-xtensa: error while loading shared libraries: libslirp.so.0: cannot open shared object file: No such file or directory

ERROR: Failed to check the tool while installed. Removing directory /home/lee/.espressif/tools/qemu-xtensa/esp_develop_9.0.0_20240606

Executing /usr/bin/python3 /home/lee/esp/esp-idf/tools/idf_tools.py install-python-env
Python 3.12.3
pip 24.3.1 from /home/lee/.espressif/python_env/idf5.5_py3.12_env/lib/python3.12/site-packages/pip (python 3.12)
Skipping the download of /home/lee/.espressif/espidf.constraints.v5.5.txt because it was downloaded recently.
Upgrading pip...
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Requirement already satisfied: pip in /home/lee/.espressif/python_env/idf5.5_py3.12_env/lib/python3.12/site-packages (24.3.1)
Upgrading setuptools...
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Requirement already satisfied: setuptools in /home/lee/.espressif/python_env/idf5.5_py3.12_env/lib/python3.12/site-packages (71.0.0)
Installing Python packages
 Constraint file: /home/lee/.espressif/espidf.constraints.v5.5.txt
 Requirement files:
  - /home/lee/esp/esp-idf/tools/requirements/requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi, https://dl.espressif.com/pypi
Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
Requirement already satisfied: setuptools in /home/lee/.espressif/python_env/idf5.5_py3.12_env/lib/python3.12/site-packages (from -r /home/lee/esp/esp-idf/tools/requirements/requirements.core.txt (line 7)) (71.0.0)
Requirement already satisfied: packaging in /home/lee/.espressif/python_env/idf5.5_py3.12_env/lib/python3.12/site-packages (from -r /home/lee/esp/esp-idf/tools/requirements/requirements.core.txt (line 8)) (24.2)
Requirement already satisfied: click in /home/lee/.espressif/python_env/idf5.5_py3.12_env/lib/python3.12/site-packages (from -r /home/lee/esp/esp-idf/tools/requirements/requirements.core.txt (line 11)) (8.1.8)
Requirement already satisfied: pyserial in /home/lee/.espressif/python_env/idf5.5_py3.12_env/lib/python3.12/site-packages (from -r /home/lee/esp/esp-idf/tools/requirements/requirements.core.txt (line 12)) (3.5)
Requirement already satisfied: cryptography in /home/lee/.espressif/python_env/idf5.5_py3.12_env/lib/python3.12/site-packages (from -r /home/lee/esp/esp-idf/tools/requirements/requirements.core.txt (line 13)) (44.0.0)
Requirement already satisfied: pyparsing in /home/lee/.espressif/python_env/idf5.5_py3.12_env/lib/python3.12/site-packages (from -r /home/lee/esp/esp-idf/tools/requirements/requirements.core.txt (line 14)) (3.2.1)
Requirement already satisfied: pyelftools in /home/lee/.espressif/python_env/idf5.5_py3.12_env/lib/python3.12/site-packages (from -r /home/lee/esp/esp-idf/tools/requirements/requirements.core.txt (line 15)) (0.31)
[... cut ...]
Requirement already satisfied: ruamel.yaml.clib>=0.2.7 in /home/lee/.espressif/python_env/idf5.5_py3.12_env/lib/python3.12/site-packages (from ruamel.yaml->idf-component-manager->-r /home/lee/esp/esp-idf/tools/requirements/requirements.core.txt (line 16)) (0.2.12)

Executing /usr/bin/python3 /home/lee/esp/esp-idf/tools/idf_tools.py export --format=key-value
Executing null/bin/python /home/lee/esp/esp-idf/tools/idf.py --version
Executing null/bin/python /home/lee/esp/esp-idf/tools/idf.py --list-targets

`

sigmaaa, 2025-01-22:
> Executing /usr/bin/python3 /home/lee/esp/esp-idf/tools/idf_tools.py export --format=key-value
Executing null/bin/python /home/lee/esp/esp-idf/tools/idf.py --version
Executing null/bin/python /home/lee/esp/esp-idf/tools/idf.py --list-targets

Hi @leenowell,

This issue occurs when the IDF_PYTHON_ENV_PATH variable is set in the system environment. We are addressing this, and a fix will be included in the next release.

In the meantime, please remove the IDF_PYTHON_ENV_PATH (it's recommended to remove all ESP-IDF-related variables to avoid conflicts) from the system environment variables. Then, restart your IDE and reinstall the tools.

leenowell, 2025-01-22:
Hi @sigmaaa 

Yes that fixed it thanks.  I run the same hello_world test and it seems to work fine on EspressifIDE.  There were a couple of other errors though so have attached the log in case it helps.

[EspressifIDE_error_log.txt](https://github.com/user-attachments/files/18502748/EspressifIDE_error_log.txt)

sigmaaa, 2025-01-22:
Hi @leenowell,

After quickly looking into the logs, I don't see any critical issues. Most of them probably happened before the tools were installed.