espressif/idf-eclipse-plugin #1121: Espressif IDE created an invalid .clang-format file

emaayan, 2025-01-20:
**Describe the bug**
espressif ide new idf project wizards creates an invalid .clang-format file 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to file ->new 
2. Click on new esp idf project 
3. create the minimal project 

.clang-format file will contain the parameter PackConstructorInitializers: NextLineOnly if you attempt to use file with clang-format manually you will see an error on this parameter. 
```

# We'll use defaults from the LLVM style, but with some modifications so that it's close to the CDT K&R style.
BasedOnStyle: LLVM
UseTab: Always
IndentWidth: 4
TabWidth: 10
PackConstructorInitializers: NextLineOnly
BreakConstructorInitializers: AfterColon
IndentAccessModifiers: false
AccessModifierOffset: -4
ColumnLimit: 200
```

**Expected behavior**
clang-formatting rules being applied after using  format source on eclipse 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Espressif-IDE Product Information:**
*** Date: יום שני, 20 בינואר 2025 בשעה 11:02:15 שעון ישראל (חורף)

*** Platform Details:

[image]

sigmaaa, 2025-01-21:
Hi @emaayan,

Thank you for reporting this. We will replace it with the default .clang-format file from the cdt lsp editor plugin

emaayan, 2025-01-21:
> Hi [[image]@emaayan](https://github.com/emaayan),
> 
> Thank you for reporting this. We will replace it with the default .clang-format file from the cdt lsp editor plugin

you might want public something about known issue, in release notes about existing configurations that may contain invalid clangd-files they are gonna say they'll fix validations https://github.com/eclipse-cdt/cdt-lsp/issues/348#issuecomment-2602039753 

i'm over trying to actually fight with clang-format on how to make it NOT join lines on structs.

closed 2025-02-19 by kolipakakondal