espressif/idf-eclipse-plugin #1126: (Using Eclipse IDE) Every time the project is run or built, the project has to build from scratch. (IDFGH-14500)

GeneralZoditch, 2025-01-24:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.3.1-dirty

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I expect the first build to take time, even I agree on that. However, as soon as I checked for updates on the Eclipse IDE, I started building my project (where I am simply making a general_gpio_output component to simplify gpio_outputs, then using it to blink an LED). Before the update, the second and consecutive builds took like 10 seconds maximum. Please help me

### What is the actual behavior?

Eveytime I changed only the pin number, the build restarted from scratch, like the build time is 2 minutes everytime

### Steps to reproduce.

N/A

### Build or installation Logs.

```plain

```

sigmaaa, 2025-01-24:
Hi @GeneralZoditch 

Thank you for reporting this. We are already addressing a similar issue, and the fix will be included in the next release. In the meantime, could you please try the artifacts linked below this page https://github.com/espressif/idf-eclipse-plugin/actions/runs/12691399517 from the build on the related PR to check if that resolves your issue? 

Let me know if you have any questions! Thank you

GeneralZoditch, 2025-01-24:
Thanks a lot! I genuinely appreciate it. You also helped me with your post
earlier about the initialization of the C++ server stuck in eclipse.