espressif/idf-eclipse-plugin #1128: Problems can become out of sync in the editor

emaayan, 2025-02-02:
eclispe cdt on #397 requested i would open this defect here .

at certain times you could see problems appearing even though they shouldn't appear but appear to be out of sync
in the example below you can see the problems description thinks i didn't type version all the way but i did. (and this is after i've hit save)

[image]

**Describe the bug**
the problem happens inherently at random times.
[image]

**Expected behavior**
that the problems view should always be clean in the event they are no errors.

kolipakakondal, 2025-02-03:
Hi @emaayan Do you see any specific scenario where this is happening, or any reproducible steps? If you have any, it would be helpful to troubleshoot the problem.

emaayan, 2025-02-03:
> Hi [[image]@emaayan](https://github.com/emaayan) Do you see any specific scenario where this is happening, or any reproducible steps? If you have any, it would be helpful to troubleshoot the problem.

the errors seems to happen with a i do some intensive save-type-save, it's not exactly reproducible, but it happens a lot,
here are a few examples: 
it seems also related to the size of the project and the file, (like for a simple hello project, it won't happen) 

[image]

[image]

[image]