espressif/idf-eclipse-plugin #1132: Avoid poothon instances overbloat on selecting ESP-IDF instance, justdrop it in favor of java. At least reuse the provided one instead

powerbroker, 2025-02-05:
**Describe the bug**
Selecting pre-installed ESP-IDF in Eclipse plugin surprisingly starts downloading and installation.
Muzz just suck-in the pre-installed instance of everything: ESP-IDF itself, build tools, etc.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Espressif/ESP-IDF Manager'
2. Click on 'Add ESP-IDF'
3. Select 'Use an existent ESP-IDF directory from file system'
4. Select proper path at 'Choose existing ESP-IDF directory:'
5. Configure Git and Poothon executables path.
6. Click 'Finish'

7. Enjoy installation process begins and goes on. 

**Expected behavior**
The provided available instance of ESP-IDF together with provided paths to executables are sucked in and reused.

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)

OS: Windows_NT
COMPUTERNAME: BLACKBOX
IDF_PATH: E:\ESP32\ESP-IDF\esp-idf-v5.2.1
KICAD_CONFIG_HOME: e:\!IDE\UserHome\KiCad
COMMANDER_INSTANCE: 1
PROCESSOR_REVISION: 7100
CommonProgramW6432: C:\Program Files\Common Files
ComSpec: C:\Windows\system32\cmd.exe
STM32CubeMX_PATH: e:\STM32\STM32CubeMX
IDF_MAINTAINER: 1
=D:: D:\
SystemRoot: C:\Windows
TEMP: d:\IDIOT~1.USE\KDV\AppData\Local\Temp
HOMEDRIVE: d:
USERPROFILE: d:\idIoT.Users\KDV
TMP: d:\IDIOT~1.USE\KDV\AppData\Local\Temp
CommonProgramFiles(x86): C:\Program Files (x86)\Common Files
NUMBER_OF_PROCESSORS: 32
IDF_PYTHON_ENV_PATH: d:\idIoT.Users\KDV\.espressif\python_env\idf5.2_py3.9_env
=Y:: Y:\
ESP_ROM_ELF_DIR: d:\idIoT.Users\KDV\.espressif\tools\esp-rom-elfs\20230320\

Operating System: windows 11
Java Runtime Version: 23.0.2+7-58
Eclipse Version: 4.34.0.v20241120-1800
Eclipse CDT Version: 11.6.1.202406241230
IDF Eclipse Plugin Version: 3.2.0.202501110717
The following Python requirements are not satisfied:

Requirement 'esptool~=4.8' was not met. Installed version: 4.7.0

To install the missing packages, please run "install.bat"

Diagnostic information:
    IDF_PYTHON_ENV_PATH: d:\idIoT.Users\KDV\.espressif\python_env\idf5.2_py3.9_env
    Python interpreter used: d:\idIoT.Users\KDV\.espressif\python_env\idf5.2_py3.9_env\Scripts\python.exe
Constraint file: d:\idIoT.Users\KDV\.espressif\espidf.constraints.v5.2.txt

Requirement files:
 - E:\ESP32\ESP-IDF\esp-idf-v5.2.1\tools\requirements\requirements.core.txt
Python being checked: d:\idIoT.Users\KDV\.espressif\python_env\idf5.2_py3.9_env\Scripts\python.exe

ESP-IDF v5.2.1-dirty
Python set for IDF_PYTHON_ENV: Python 3.11.5

powerbroker, 2025-02-10:
digging into the issue brings great result to us: re-aiming ESP-IDF onto it's internal poothon(.espressif\python_env\idf5.4_py3.11_env\Scripts\...) makes plugin absolutely unuseable and potentially leads to system crash, since poothon process spawns itself forever:

[image]

alirana01, 2025-02-10:
@powerbroker we are sorry that you are facing these issues and we really appreciate all the contributions in regard to our **OpenSource** plugin no matter if they even come in form of issues it helps us diagnose and improve. That being said there are certain limitations to it as well. We are continuously working in improving user experience and for this reason we will soon be moving towards the [EIM](https://github.com/espressif/idf-im-cli) in future.

All that aside we would be glad to help you out and dig deeper in the issue with you over a call if with suitable time you have.

You can reach out to us here and we will schedule and analyze the issue you are facing
ide.support@espressif.com