espressif/idf-eclipse-plugin #1148: Avoid creating '.clang-format' on each compilation

carlessole, 2025-02-11:
After installing the latest Espresif-IDE version, each time I compile the project, a '.clang-format' file is generated.

I need to disable this autogeneration as I am currently using one from my root directory.

Thanks,

Carles

sigmaaa, 2025-02-11:
Hi @carlessole,

Thank you for reporting this. We generate the .clang-format file after building the project if it is missing to ensure formatting is available for users migrating from the non-LSP plugin version. However, you’ve brought up a case I hadn’t considered, and I agree that this option could be useful. The best place for it might be on the Espressif preferences page. What do you think?

[image]

carlessole, 2025-02-11:
Thanks for reaching out so soon @sigmaaa ,
Yes, for me it is a perfect place.

Another good place could be from 'Preferences>C/C++>Editor(LSP)>clangd?

[image]

So, for me, place it where you prefer :),

Thanks

closed 2025-02-26 by kolipakakondal