espressif/idf-eclipse-plugin #1167: ESP IDE - Ctrl click or F3 not working

Chaithra-SE, 2025-02-26:
I am facing issues with ESP-IDE when a project is imported and trying to open the declaration of any function or variable. I checked the error logs and here is the message. I tried indexing also, but no improvement.

Attached screenshot of the issue. Pl help at the earliest.

[image]

kolipakakondal, 2025-02-26:
Hello @Chaithra-SE Could you share the complete error log(export and attach the file here) and the clangd configuration path on your system? The clangd configuration should be something like this, and it is auto-configured.

[image]

kolipakakondal, 2025-02-27:
>java.util.concurrent.CompletionException: java.nio.file.InvalidPathException: Illegal char <"> at index 0: "C:\Users\SESA423801\AppData\Local\Programs\Python\Launcher\

While we check further, my quick observation is that your PATH has double quotes around it. Could you navigate to IDE Preferences > C/C++ > Build > Environment, look for the PATH, copy it into Notepad, check if it contains double quotes, remove them, and save the changes?

[image]

Chaithra-SE, 2025-02-27:
Hello @kolipakakondal ,
Many thanks for the reply. I am able to go to definition now. The issue was caused by double quotes in the PATH. 

Thanks a lot & Have a great day ahead !!

closed 2025-02-27 by Chaithra-SE