espressif/idf-eclipse-plugin #1172: "Installing tools..." problem: trying to add: ESP-IDF

georgewicks, 2025-03-06:
I have tried to add ESP-IDF using an already downloaded ESP-IDF (5.3.1), and also tried with having the plug-in install a version via plug-in download, and it fails in both cases
(environment: Windows 10)

Error log:
Traceback (most recent call last):
  File "W:\Development\esp-idf-v5.3.2\tools\idf_tools.py", line 3239, in <module>
    main(sys.argv[1:])
  File "W:\Development\esp-idf-v5.3.2\tools\idf_tools.py", line 3231, in main
    action_func(args)
  File "W:\Development\esp-idf-v5.3.2\tools\idf_tools.py", line 2154, in action_export
    tools_info = load_tools_info()
  File "W:\Development\esp-idf-v5.3.2\tools\idf_tools.py", line 1610, in load_tools_info
    return parse_tools_info_json(tools_info)  # type: ignore
  File "W:\Development\esp-idf-v5.3.2\tools\idf_tools.py", line 1625, in parse_tools_info_json
    tool = IDFTool.from_json(tool_dict)
  File "W:\Development\esp-idf-v5.3.2\tools\idf_tools.py", line 1239, in from_json
    raise RuntimeError(f'supported_targets for tool {tool_name} is not a list of strings')
RuntimeError: supported_targets for tool xtensa-esp32-elf is not a list of strings

sigmaaa, 2025-03-13:
Hi @georgewicks,

Could you please try the latest release and check if the problem occurs there? [Link](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.3.0)

FabITA1, 2025-03-14:
I have tried with Espressif-IDE 3.3.0 installing IDF5.1 and I have similar problem:

Traceback (most recent call last):
  File "C:\Espressif\ESP-IDF\esp-idf-v5.1\tools\idf_tools.py", line 2831, in <module>
    main(sys.argv[1:])
  File "C:\Espressif\ESP-IDF\esp-idf-v5.1\tools\idf_tools.py", line 2823, in main
    action_func(args)
  File "C:\Espressif\ESP-IDF\esp-idf-v5.1\tools\idf_tools.py", line 1807, in action_export
    tools_info = load_tools_info()
                 ^^^^^^^^^^^^^^^^^
  File "C:\Espressif\ESP-IDF\esp-idf-v5.1\tools\idf_tools.py", line 1344, in load_tools_info
    return parse_tools_info_json(tools_info)  # type: ignore
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Espressif\ESP-IDF\esp-idf-v5.1\tools\idf_tools.py", line 1362, in parse_tools_info_json
    tool = IDFTool.from_json(tool_dict)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Espressif\ESP-IDF\esp-idf-v5.1\tools\idf_tools.py", line 1049, in from_json
    if Platforms.get(platform_id) is None:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Espressif\ESP-IDF\esp-idf-v5.1\tools\idf_tools.py", line 211, in get
    platform_name = Platforms.detect_linux_arm_platform(platform_name)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Espressif\ESP-IDF\esp-idf-v5.1\tools\idf_tools.py", line 185, in detect_linux_arm_platform
    with open(sys.executable, 'rb') as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument: 'C:\\Users\\abcdefg\\AppData\\Local\\Microsoft\\WindowsApps\\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\\python.exe'

sigmaaa, 2025-03-17:
Hi @FabITA1,

I see you're using Python from the Microsoft Store. Which path did you select for the Python executable when installing the tools? From the error, I see it could be 'C:\Users\abcdefg\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\python.exe', which is a stub that redirects execution. Make sure that the path looks like this for Microsoft Python

[image]

If that doesn't help, I recommend installing Python from the official source: [python.org](https://www.python.org/downloads).

FabITA1, 2025-03-17:
As I said, I installed a new enviroment. This  installed the Espressif-IDE 3.1.0 and it works fine. When upgrade to 3.3.0 with "check for update" menu, I obtain many errors of the type attached.

[image]

sigmaaa, 2025-03-17:
> As I said, I installed a new enviroment. This installed the Espressif-IDE 3.1.0 and it works fine. When upgrade to 3.3.0 with "check for update" menu, I obtain many errors of the type attached.
> 
> [image]

It seems that the issue you're encountering might be related to the latest Eclipse CDT 12. Have you also updated both Eclipse and the CDT plugin? Currently, we support Eclipse version 24-09 and CDT version 11.6.0. We are working on supporting the latest version, and it will be available in the next release.

To avoid compatibility issues, I recommend downloading Espressif-IDE from [this link](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.3.0). Alternatively, you can install the Espressif plugin on top of Eclipse CDT 11.6 downloaded from [here](https://projects.eclipse.org/projects/tools.cdt/releases/11.6.0).

sigmaaa, 2025-03-17:
> It seems that the issue you're encountering might be related to the latest Eclipse CDT 12. Have you also updated both Eclipse and the CDT plugin? Currently, we support Eclipse version 24-09 and CDT [...]

If updating the CDT plugin is the issue, it's possible to revert the update via the `Installation History` : Help -> About Eclipse IDE -> Installation Details -> Revert

[image]

FabITA1, 2025-03-17:
I have installed the tool using the tool download from here: https://dl.espressif.com/dl/esp-idf/
This tool install the Espressif-IDE 3.1.0. Then, from the Espressif-IDE, I updeted all it proposed from "Check for update" command.
Now I have download the only Espressif-IDE from the link you suggest and replaced manually the old 3.1.0: this seems to works. 
I'll do some more checks.
Tanks @sigmaaa

closed 2026-04-17 by AndriiFilippov