espressif/idf-eclipse-plugin #1176: bootloader.elf: 1: Syntax error: end of file unexpected (IDFGH-14705)

Axa-Ru, 2025-03-11:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v 5.4

### Operating System used.

Linux

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Expected flash to esp32 MCU

### What is the actual behavior?

In Esp-IDF the project is built without errors and is successfully flashed.
In Eclipse the project is built without errors.

esp32-ttg0-display.bin binary size 0x331d0 bytes. Smallest app partition is 0x100000 bytes. 0xcce30 bytes (80%) free.
[100%] Built target app_check_size
[100%] Built target app
Build complete (0 errors, 0 warnings): /home/axa/eclipse-wrkspace/esp32-ttgo-display/build/default

At the flashing stage it gives an error

/home/axa/eclipse-wrkspace/esp32-ttgo-display/build/default/bootloader/bootloader.elf: 1: Syntax error: end of file unexpected

### Steps to reproduce.

1. In Eclipse IDE Run->Run

### Build or installation Logs.

```plain

```

### More Information.

Operating system Linux Mint.
IDF-Esp 5.4
Eclipse 4.34.0

kolipakakondal, 2025-03-18:
Hello,

> /home/axa/eclipse-wrkspace/esp32-ttgo-display/build/default/bootloader/bootloader.elf

Pointing to the wrong build directory: /build/default/. I believe it should be /build/bootloader/bootloader.elf instead.

Please check if you have configured a custom build directory. By default, it should be project/build, but in your case, it is showing /build/default/.

You can also set a custom build directory name in the configuration page.

[image]

Indastri, 2025-08-05:
Since there was no further response from the user - closing. If you will encounter more problems feel free to reopen this or create a new issue.

Cheers!