espressif/idf-eclipse-plugin #1186: CMake Error: Unable to Find project.cmake in ESP-IDF Despite Environment Setup (IDFGH-14864)

Prroyal, 2025-03-17:
While configuring my ESP32 project in Espressif IDE, CMake fails with an error stating that it cannot find project.cmake at null:/home/amantya/esp/esp-idf/tools/cmake/project.cmake. The toolchain is installed, IDF_PATH is set, and the build works in the terminal but fails in the IDE. Additionally, ninja: error: loading 'build.ninja': No such file or directory occurs.

sigmaaa, 2025-03-17:
Hi @Prroyal,

What version of Espressif-IDE are you using? If it's not 3.3.0, could you please try installing it to see if the problem occurs there? [Espressif-IDE v3.3.0](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.3.0)
Thank you!

Prroyal, 2025-03-17:
The version of Espressif-IDE being used is 3.3.0.

alirana01, 2025-03-17:
@Prroyal can you please share the logs and some more information

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

Prroyal, 2025-03-17:
@alirana01 

[error.log](https://github.com/user-attachments/files/19284348/error.log)
[product_info.log](https://github.com/user-attachments/files/19284349/product_info.log)

alirana01, 2025-03-17:
Hi @Prroyal 
The logs are seemingly showing tools installation issue inside the IDE. 
Can you please see that you switch to a new workspace and install the tools in IDE via ESP-IDF Manager option.
If you run into issues there please share the logs.
Also kindly share which version of ESP-IDF are you using?

[image]

Prroyal, 2025-03-17:
@alirana01 The version of Espressif-IDE being used is 3.3.0

alirana01, 2025-03-17:
> [@alirana01](https://github.com/alirana01) The version of Espressif-IDE being used is 3.3.0

Yes but which ESP-IDF framework are you using?

alirana01, 2025-03-17:
Hi @Prroyal 
Thank you for the information. If you have not installed the tools via IDE please try to do so. [Here](https://docs.espressif.com/projects/espressif-ide/en/latest/installation.html#esp-idf-and-tools-installation) is the documentation for that.

Further I see that cmake is probably missing from the environment.

Can you please see what is the output of this command on your terminal 
`which cmake`

Also please share the PATH variable value from the IDE setting.

[image]

Prroyal, 2025-03-17:
@alirana01 cmake error is solved  but project is giving semantic errors

alirana01, 2025-03-17:
@Prroyal thank you for the update can you please create another issue on our issues [page](https://github.com/espressif/idf-eclipse-plugin/issues) for the semantic errors, I am closing this one as cmake issue is resolved

closed 2025-03-17 by alirana01