espressif/idf-eclipse-plugin #1189: Semantic Errors in Espressif IDE - Unresolved Identifiers (CONFIG_BLINK_PERIOD, portTICK_PERIOD_MS, uint8_t) (IDFGH-14875)

Prroyal, 2025-03-18:
I am facing semantic errors in Espressif IDE while working on a simple blink example. The IDE is unable to resolve CONFIG_BLINK_PERIOD, portTICK_PERIOD_MS, and uint8_t, even though the project builds successfully using idf.py build.

sigmaaa, 2025-03-18:
Hi @Prroyal,

If you're using Espressif-IDE, there's no need to build the project with `idf.py build` you can simply use the build button in the IDE. More details are in the official documentation: [Espressif-IDE Build Project](https://docs.espressif.com/projects/espressif-ide/en/latest/buildproject.html).

Even if the project is built using the `idf.py build` command, it's recommended to use the build button at least once to automatically configure Clangd options for a current project. This ensures proper functionality, such as:

[image]

Indastri, 2025-08-13:
Since there was no further response from the user side - closing. If you encounter further problems feel free to reopen this or create a new issue.

Cheers!