espressif/idf-eclipse-plugin #1191: Inactive code block not working

mmrein, 2025-03-25:
### Describe the bug

Apart that LSP editor makes inactive code block look like comment, which looses its syntax higlhight and therefore makes multi-target builds much harder, **it can even be straight up wrong**.

### Screenshots

I made myself some core affinity setup indication so I can see it at one place without needing to run sdkconfig and searching through various components.

This is how it looked in IDE v2.12 - you can see which configuration is active and which is not:

[image]

And this is same setup with IDE v3.3:

[image]

### Steps to reproduce

You can try adding this or simillar code to you project and build it:

```c
/* ESP System Settings / Main task core affinity */
#if defined CONFIG_ESP_MAIN_TASK_AFFINITY_NO_AFFINITY	// Main task has affinity
#elif defined CONFIG_ESP_MAIN_TASK_AFFINITY_CPU0		// Main task set to Core 0
#elif defined CONFIG_ESP_MAIN_TASK_AFFINITY_CPU1		// Main task set to Core 1
#endif
/* WiFi / WiFi task */		
#if defined CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_0		// WiFi task set to Core 0
#elif defined CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_1	// WiFi task set to Core 1
#endif
/* MDF Common / MDF task */
#if defined CONFIG_MDF_PINNED_TO_CORE_0					// MDF task set to Core 0
#elif defined CONFIG_MDF_PINNED_TO_CORE_1				// MDF task set to Core 1
#endif
/* LWIP / TCPIP task affinity */
#if defined CONFIG_LWIP_TCPIP_TASK_AFFINITY_NO_AFFINITY	// TCPIP task has affinity
#elif defined CONFIG_LWIP_TCPIP_TASK_AFFINITY_CPU0		// TCPIP task set to Core 0
#elif defined CONFIG_LWIP_TCPIP_TASK_AFFINITY_CPU1		// TCPIP task set to Core 1
#endif
/* mDNS task - may be used with webpages */
#if defined CONFIG_MDNS_TASK_AFFINITY_NO_AFFINITY		// mDNS task has affinity
#elif defined CONFIG_MDNS_TASK_AFFINITY_CPU0			// mDNS task set to Core 0
#elif defined CONFIG_MDNS_TASK_AFFINITY_CPU1			// mDNS task set to Core 1
#endif
```

### Expected behavior

Clear and correct distinction between active and inactive code block, preferably by change of background color of whole block **while keeping original syntax highlight**.

**Espressif-IDE Product Information:**

[espprodinf.txt](https://github.com/user-attachments/files/19446453/espprodinf.txt)

mmrein, 2025-03-25:
**Might as well address some more points:**
- All three bot recomendations were done in the first place
- I am still using IDE 2.12 with IDF v4.4 in production, but I need to switch to IDF v5+ finally as there are some major security issues addressed there (WPA3 for SoftAP/mesh mainly)
- I tried using VSCode with clangd extension which would solve this issue as expected, but it unfortunatelly reports tons of false errors and is unusable for that reason. Tried reporting it in https://github.com/espressif/vscode-esp-idf-extension/issues/1352, but that went nowhere and just autoclosed (in 14 days...) for inactivity.

sigmaaa, 2026-08-28:
Closing this since the problem is related to the LSP editor itself.

closed 2026-08-28 by sigmaaa