espressif/idf-eclipse-plugin #1197: non working CDT-LDP after installation of IDF-Eclipse-Plugin

mzimail, 2025-04-03:
**Describe the bug**
As soon as ESP-IDF-Eclipse Plugin is installed, other projects using CDT-LSP (standard C/C++ project using editor with clangd language server) are not working any more: Language Server (clangd-19) is not started.
This applies to existing and new C/C++ projects, regardless of there are any ESP-IDF projects in the same workspace or not. Simple installation of ESP-IDF-Eclipse breaks previously working projects of those other natures.

**To Reproduce**
Steps to reproduce the behavior:
1. Create blank Workspace
2. Click on 'File -> New -> C/C++ Project'
3. Choose "CMake Project"
4. Click "Next"
5. Assign Project Name "x86helloworld"
6. Click "Finish"
7. Open "x86helloworld.cpp" using "C/C++ Editor (LSP)"
8. Open "Windows -> Whow View -> Other -> Language Servers -> Language Servers"

**Expected behavior**
The configured language server (/usr/bin/clangd-19 in this case) should be started just at is if, when ESP-IDF-plugin is not installed.
Syntax errors should be highlighted in editor accordlingly, but they aren't.

**Screenshots**

[image]

**Espressif-IDE Product Information:**
[idfproductinfo.txt](https://github.com/user-attachments/files/19594121/idfproductinfo.txt)

**Eclipse Error log:** 
[errorlog.log](https://github.com/user-attachments/files/19594112/errorlog.log)

**

sigmaaa, 2025-04-04:
Hi @mzimail 

Could you please check the Clangd preferences page?

[image]

mzimail, 2025-04-04:
Hi @sigmaaa,
thanks for having a closer look at my problem.

The LSP Preferences page looks like this, when in an Workspace with ESP-IDF projects, according clangd from ESP-IDF is then starting as expected:

[image]

The LSP Preferences page looks like this, when in an Workspace with pure C/C++ projects, then clangd is not starting as soon as IDF-eclipse-plugin is installed, but clangd would start using the same settings as soon as  IDF-eclipse-plugin is not installed:

[image]

So Installation of the plugin effectively breaks projects / configurations that used to work previously.