espressif/idf-eclipse-plugin #1207: Eclipse 2025-03(4.35.0) compatibility issues

GooogLeMorons, 2025-04-26:
**Describe the bug**
many internal errors during "Change build Configuration"

**To Reproduce**
Install Eclipse C/C++ Embedded
Install Espressif plugin
Add ESP-IDF
Create simple Espressif project

enjoy tons of:

An internal error occurred during: "Change Build Configurations".
Receiver class com.espressif.idf.core.build.IDFBuildConfigurationProvider does not define or inherit an implementation of the resolved method 'abstract org.eclipse.cdt.core.build.ICBuildConfiguration createCBuildConfiguration(org.eclipse.core.resources.IProject, org.eclipse.cdt.core.build.IToolChain, java.lang.String, org.eclipse.launchbar.core.target.ILaunchTarget, org.eclipse.core.runtime.IProgressMonitor)' of interface org.eclipse.cdt.core.build.ICBuildConfigurationProvider.

**Expected behavior**
The project is created and built successfully

**Screenshots**
[image]

**Espressif-IDE Product Information:**
Operating System: windows 11
Java Runtime Version: 24+36-3646
Eclipse Version: 4.35.0.v20250228-0140
Eclipse CDT Version: 12.0.0.202501171709
IDF Eclipse Plugin Version: 3.3.0.202503071213
ESP-IDF v5.5-dev-1860-g0d6099ec53
Python set for IDF_PYTHON_ENV: Python 3.11.5

**Eclipse Error log:** 
[eclipse-errors.log](https://github.com/user-attachments/files/19920456/eclipse-errors.log)

sigmaaa, 2025-04-26:
Hi @GooogLeMorons, 

CDT 12.0 is not supported on version 3.3.0. However, support will be available in version 3.4.0, which is currently in the final testing stage and will be released soon.

If you'd like early access, you can download the update site or the latest Espressif IDE from the link below.
https://github.com/espressif/idf-eclipse-plugin/actions/runs/14668065172

Thank you!

apalazzi, 2025-04-28:
Is it possible to add a warning/error when trying to install the plugin on Eclipse versions that are not compatible? I went on and after installing the plugin I upgraded a workspace to the latest version, just to discover later that the plugin doesn't work correctly...

sigmaaa, 2025-04-30:
> Is it possible to add a warning/error when trying to install the plugin on Eclipse versions that are not compatible? I went on and after installing the plugin I upgraded a workspace to the latest version, just to discover later that the plugin doesn't work correctly...

Hi @apalazzi,

Sorry to hear you ran into this issue. At the moment, we can restrict the installation of dependency versions that haven’t been tested with specific versions of the IDF Eclipse plugin, but we generally avoid doing so since dependency changes rarely cause issues for our plugin.

That said, a good compromise might be to block installation when a new major version of a dependency is detected, as that's when breaking changes are more likely. 

Thanks again for the feedback!

sigmaaa, 2025-05-12:
Closing this due to a new release that addresses the issue: https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.4.0

closed 2025-05-12 by sigmaaa