espressif/idf-eclipse-plugin #1211: Error: Build is not configured correctly

apalazzi, 2025-04-29:
**Describe the bug**
When I try to build i get "Build is not configured correctly"
I tried uninstalling and reinstalling everyting - Eclipse, latest and 2024-06, the IDF plugin, the IDF SKD in multiple version, delete the project and recreating it, but the result is always the same: build is not configured correctly.
Honestly, this issue and others are becoming so frustrating that I'm considering switching to Emacs... I love Eclipse but if I have to waste days figuring out issues it's better to use something else.

**To Reproduce**
Try to build a project in Eclipse.

**Expected behavior**
The project should build

**Espressif-IDE Product Information:**
Executing /usr/bin/python3 /home/andrea/esp/esp-eclipse/esp-idf-v5.1.6/tools/idf_tools.py list
* xtensa-esp-elf-gdb: GDB for Xtensa
  - 12.1_20231023 (recommended, installed)
* riscv32-esp-elf-gdb: GDB for RISC-V
  - 12.1_20231023 (recommended, installed)
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* esp-clang: Toolchain for all Espressif chips based on clang (optional)
  - 15.0.0-23786128ae (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.35_20220830 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.30.2 (recommended, installed)
  - 3.16.3 (supported)
* openocd-esp32: OpenOCD for ESP32
  - v0.12.0-esp32-20241016 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.12.1 (recommended, installed)
* esp-rom-elfs: ESP ROM ELFs
  - 20230320 (recommended, installed)
* qemu-xtensa: QEMU for Xtensa (optional)
  - esp_develop_8.1.3_20231206 (recommended, installed)
* qemu-riscv32: QEMU for RISC-V (optional)
  - esp_develop_8.1.3_20231206 (recommended, installed)
OK

CDT Build environment variables
PATH: /home/andrea/esp/esp-eclipse/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/home/andrea/esp/esp-eclipse/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/home/andrea/esp/esp-eclipse/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/andrea/esp/esp-eclipse/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/andrea/esp/esp-eclipse/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/andrea/esp/esp-eclipse/tools/esp-clang/15.0.0-23786128ae/esp-clang/bin:/home/andrea/esp/esp-eclipse/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/andrea/esp/esp-eclipse/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/andrea/esp/esp-eclipse/tools/cmake/3.30.2/bin:/home/andrea/esp/esp-eclipse/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:/home/andrea/esp/esp-eclipse/tools/ninja/1.12.1/:/home/andrea/esp/esp-eclipse/tools/qemu-xtensa/esp_develop_8.1.3_20231206/qemu/bin:/home/andrea/esp/esp-eclipse/tools/qemu-riscv32/esp_develop_8.1.3_20231206/qemu/bin:/home/andrea/esp/esp-eclipse/python_env/idf5.1_py3.13_env/bin:/home/andrea/esp/esp-eclipse/esp-idf-v5.1.6/tools:/home/andrea/.local/bin:/home/andrea/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
XAUTHORITY: /run/user/1000/gdm/Xauthority
XMODIFIERS: @im=ibus
XDG_DATA_DIRS: /usr/share/gnome:/usr/share/gnome:/usr/local/share/:/usr/share/
GDMSESSION: cinnamon
GTK_IM_MODULE: ibus
DBUS_SESSION_BUS_ADDRESS: unix:path=/run/user/1000/bus
ESP_IDF_VERSION: 5.1.6
XDG_CURRENT_DESKTOP: X-Cinnamon
OPENOCD_SCRIPTS: /home/andrea/esp/esp-eclipse/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts
IDF_DEACTIVATE_FILE_PATH: /tmp/tmprqen9sbdidf_11056
USERNAME: andrea
SESSION_MANAGER: local/navicella:@/tmp/.ICE-unix/2509,unix/navicella:/tmp/.ICE-unix/2509
LOGNAME: andrea
PWD: /home/andrea
SHELL: /bin/bash
GIO_LAUNCHED_DESKTOP_FILE: /home/andrea/Scrivania/Eclipse IDE for Embedded C and C++ Developers - 2024-06.desktop
GNOME_DESKTOP_SESSION_ID: this-is-deprecated
IDF_COMPONENT_MANAGER: 1
GTK_MODULES: gail:atk-bridge
CLUTTER_IM_MODULE: ibus
PYTHON_EXE_PATH: /usr/bin/python3
XDG_SESSION_DESKTOP: cinnamon
OXYGEN_DISABLE_INNER_SHADOWS_HACK: 1
QT_IM_MODULE: ibus
LANG: it_IT.UTF-8
XDG_SESSION_ID: 2
XDG_SESSION_TYPE: x11
IDF_TOOLS_PATH: /home/andrea/esp/esp-eclipse
DISPLAY: :0
XDG_SESSION_CLASS: user
GDM_LANG: it_IT.UTF-8
GTK3_MODULES: xapp-gtk3-module
LIBOVERLAY_SCROLLBAR: 0
IDF_PATH: /home/andrea/esp/esp-eclipse/esp-idf-v5.1.6
GPG_AGENT_INFO: /run/user/1000/gnupg/S.gpg-agent:0:1
DESKTOP_SESSION: cinnamon
GDK_CORE_DEVICE_EVENTS: 1
USER: andrea
GIO_LAUNCHED_DESKTOP_FILE_PID: 28513
WINDOWPATH: 2
QT_ACCESSIBILITY: 1
SSH_AUTH_SOCK: /run/user/1000/keyring/ssh
XDG_SEAT: seat0
IDF_MAINTAINER: 1
XDG_RUNTIME_DIR: /run/user/1000
XDG_VTNR: 2
IDF_PYTHON_ENV_PATH: /home/andrea/esp/esp-eclipse/python_env/idf5.1_py3.13_env
HOME: /home/andrea
ESP_ROM_ELF_DIR: /home/andrea/esp/esp-eclipse/tools/esp-rom-elfs/20230320/

Operating System: linux
Java Runtime Version: 21.0.7-ea+8-Debian-1
Eclipse Version: 4.35.0.v20250228-0140
Eclipse CDT Version: 12.0.0.202501171709
IDF Eclipse Plugin Version: 3.3.0.202503071213
ESP-IDF v5.1.6-dirty
Python set for IDF_PYTHON_ENV: Python 3.13.2

[esp-idf.log](https://github.com/user-attachments/files/19966152/esp-idf.log)

**Eclipse Error log:** 
See attachment.

kolipakakondal, 2025-04-30:
Hello @apalazzi Based on the logs, you're working with the eclipse version(2024-03) and CDT version(12.0.0) which is not yet supported the IDF eclipse plugin.

We are planning to release IEP 3.4.0 this either this week or early next week to support Eclipse 2024-03 and CDT 12.0.0, if you would like to try our dev build, please download from here.

[Espressif-IDE-3.4.0-linux.gtk.x86_64](https://github.com/espressif/idf-eclipse-plugin/actions/runs/14703438505/artifacts/3020180891) (this is suitable for your system)
[Espressif-IDE-3.4.0-linux.gtk.aarch64](https://github.com/espressif/idf-eclipse-plugin/actions/runs/14703438505/artifacts/3020182044)

If you're looking for a stable version, please download it from here
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-3.3.0-linux.gtk.x86_64.tar.gz (this is suitable for your system)
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-3.3.0-linux.gtk.aarch64.tar.gz

I hope that helps and let us know how does it goes!

Thanks!

kolipakakondal, 2025-05-12:
Hello,

We have released v3.4.0. Please check it here: https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.4.0
This release addresses the issue related to Eclipse 2025-03 and CDT 3.1.0.

Closing this issue. Feel free to reopen it if you still encounter the same problem with the latest builds.

Thanks.

closed 2025-05-12 by kolipakakondal