espressif/idf-eclipse-plugin #1219: eclipse shows errors afer installation

sreevalli802, 2025-05-11:
Hi,
I have installed Eclipse Version: 2025-03 (4.35.0) as soon as it is installed, I found 16 errors without writing a single line of code. I have installed JDK 24. Can anyone help me?

sreevalli802, 2025-05-11:
Hi,
I have installed Eclipse Version: 2025-03 (4.35.0) as soon as it is installed, I found 16 errors without writing a single line of code. I have installed JDK 24. Can anyone help me?

sigmaaa, 2025-05-11:
Hi @sreevalli802,

Are you using the latest 3.4.0 release? [v3.4.0 release](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.4.0)

If not, please install version 3.4.0, as older versions are not compatible with CDT 12. To install 3.4.0, use this update site:

Installing IDF Plugins using the local archive file:
[com.espressif.idf.update-3.4.0.zip](https://dl.espressif.com/dl/idf-eclipse-plugin/updates/com.espressif.idf.update-3.4.0.zip)

Alternatively, if you'd like to avoid dealing with dependencies, you can download Espressif IDE, which includes all dependencies and uses the latest supported Eclipse IDE:

[Espressif IDE for macOS (aarch64)](https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-macosx-cocoa-aarch64-v3.4.0.dmg)

[Espressif IDE for macOS (x86_64)](https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-macosx-cocoa-x86_64-v3.4.0.dmg)

[Espressif IDE for Windows](https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-3.4.0-win32.win32.x86_64.zip)

[Espressif IDE for Linux (gtk-x86_64)](https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-3.4.0-linux.gtk.x86_64.tar.gz)

[Espressif IDE for Linux (gtk-aarch64)](https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-3.4.0-linux.gtk.aarch64.tar.gz)

Could you also please share the logs to help us investigate further?

Thank you.

kolipakakondal, 2025-06-10:
Hello,

If you’re using our latest IDE version v3.4.0, the initial CMake build will automatically run as soon as a project is created, and header resolution for LSP editors should work as expected.

Please verify and re-open the issue if you’re still experiencing the same problem with v3.4.0.

Closing this issue due to inactivity. If you encounter any other issues with the IDE, feel free to open a new one.

Thanks!

closed 2025-06-10 by kolipakakondal