espressif/idf-eclipse-plugin #1221: Espressif-IDE freezes when editing code if there are Errors or Warnings with type: 'Language Servers'

Vincent-Dalstra, 2025-05-13:
**Describe the bug**
When the 'problem' view is open, and the list of 'Warnings' or 'Errors'  is open, the application will freeze constantly. 

I don't have much experience debugging eclipse, but I've tried to narrow down when exactly this will happen:

- The duration of the freeze is equal to approximately 0.25s per Warning/Error that has Type = Language Servers. Other types of errors/warnings have no effect.
- Any input while the application is frozen takes effect after the freeze ends.
- Freezes will not occur while the 'Errors' and 'Warnings' lists are collapsed.
  - If a list is collapsed, then immediately un-collapsed, no freeze occurs.
  - If a list is collapsed, and one of the actions below is taken, then attempting to un-collapse the list will cause a freeze.
- A freeze will occur within about 0.5s after:
  - Editing a C/C++ file in any way (e.g. undo/redo and copy-paste)
  - Opening a C/C++ file.
- This only happens when editing C/C++ files (.c, .h, .cpp, .hpp). Editing CMakelists.txt and KConfig.projbuild will NOT trigger freezes.

When I run espressif-ide from a terminal, I get the following output during a freeze. The number of lines is = 2x the number of Errors/Warnings with Type = Language Servers.

e.g 4 warnings with Type = Language Servers:

```
I[14:55:22.264] <-- textDocument/codeAction("19")
I[14:55:22.264] --> reply:textDocument/codeAction("19") 0 ms
I[14:55:22.524] <-- textDocument/codeAction("20")
I[14:55:22.524] --> reply:textDocument/codeAction("20") 0 ms
I[14:55:22.786] <-- textDocument/codeAction("21")
I[14:55:22.786] --> reply:textDocument/codeAction("21") 0 ms
I[14:55:23.048] <-- textDocument/codeAction("22")
I[14:55:23.048] --> reply:textDocument/codeAction("22") 0 ms
```

**To Reproduce**
Steps to reproduce the behavior:

1. Open and build a project that has a 'Warning' or 'Error' with Type = Language Servers.
2. Open a source file (this does not have to be the same file as the warning or error).
3. Open the 'Problems' view
4. Un-collapse 'Warnings' or 'Errors'.
At this point, the IDE will freeze for a period of time, depending on how many errors / warnings there are (~0.25s x the number). Only after that will the list of errors/warnings be opened.

5. Type some text into a source file
The freeze will occur again, for the same amount of time. Any letters typed
6. Collapse 'Warning' and 'Error'
7. Add or remove some text from the source file
8. No freeze will occur

**Expected behavior**
The application should not freeze when editing source files, simply because the list of Warnings or Errors is visible.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)
**Done**

[Espressif-IDE Product Information.txt](https://github.com/user-attachments/files/20182857/Espressif-IDE.Product.Information.txt)

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

Vincent-Dalstra, 2025-05-13:
1. Disabled the CDT indexer. Problem persists.
2. Clangd configuration is correct: matches the [referenced page](https://docs.espressif.com/projects/espressif-ide/en/latest/additionalfeatures/clangd_cdt_support.html) with the only difference being newer versions:
  path = `/home/vincent/.espressif/tools/esp-clang/esp-18.1.2_20240912/esp-clang/bin/clangd`
  additional = `--compile-commands-dir=/home/vincent/eclipse-workspace/room_audio_player/build`
3. Already high: -Xmx4096m

Vincent-Dalstra, 2025-05-14:
I'd like to investigate this further and try and find a solution, but don't really know how to debug eclipse or LSP's.

In particular, if anyone can provide some instructions on how to get more verbose output from the terminal running the IDE, that would be very much appreciated. Currently I can only see LOGI type messages. I expect that LOGD would have more detail about what is actually happening...

kolipakakondal, 2025-05-14:
Hi @Vincent-Dalstra It seems that when opening the Problems view with errors and markers, the IDE is making multiple textDocument/publishDiagnostics and textDocument/codeAction calls, which may be contributing to the performance issues.

I don’t have access to a Linux environment at the moment, so I tried testing this quickly on macOS. I did notice some lag when expanding the Problems view, but it doesn’t appear to impact the overall IDE experience significantly.

Ok, will verify this behavior on Linux as well and see what we can do about it.

Regarding the enabling the verbose for LSP, you can enable "**Log to console**" option in t**he Editor(LSP) > Clangd preferences**.