espressif/idf-eclipse-plugin #1227: Internal error occurred during: "Update .clangd" on 3.4.0

Mikl604, 2025-05-19:
Hi.
Updated to 3.4.0, now all projects give the following message when compiled: 

"An internal error occurred during: "Update .clangd".
'void org.eclipse.core.resources.IFile.setContents(byte[], int, org.eclipse.core.runtime.IProgressMonitor)'".

The projects are completely cleaned, compiled without errors, but this message still pops up.

Mikl604, 2025-05-19:
It doesn't help. My .clangd:

CompileFlags:
  CompilationDatabase: C:\Espressif\Projects\StendPTZ_new\build
  Remove: [-m*, -f*]

[image]

sigmaaa, 2025-05-19:
Hi @Mikl604,

I wasn’t able to reproduce the issue on my end, but it appears to be related to the LSP CDT plugin, as our plugin does not modify the .clangd file.

Could you please share the full log to help us investigate further? You can refer to the [error log guide](https://docs.espressif.com/projects/espressif-ide/en/latest/troubleshooting.html#error-log) for more details on how to obtain it.

Additionally, it’s worth noting that our plugin automatically sets the `--compile-commands-dir`, so indexing does not rely on the CompilationDatabase flag in the .clangd file. If enabling the "Set compilation database path in .clangd file" option is causing issues, I think it's safe to disable it. (Preferences -> C/C++ -> Editor (LSP) -> clangd ->  disable `Set compilation database path in .clangd file`

Let me know if that resolves the problem for you. Thanks!

Mikl604, 2025-05-20:
Disable Set compilation database path in .clangd file, It didn't help.

Mikl604, 2025-05-20:
here is the error log

[err.log](https://github.com/user-attachments/files/20326279/err.log)

sigmaaa, 2025-05-20:
Hi @Mikl604,

Thanks for the logs. It seems that the `org.eclipse.core.resources` plugin is an older version than required. The method ClangdCompilationDatabaseSetterBase.writeClangdConfigFile from the CDT LSP uses IFile.setContents(byte[], int, org.eclipse.core.runtime.IProgressMonitor) and IFile.create(byte[], boolean, boolean, org.eclipse.core.runtime.IProgressMonitor), which are available starting from `org.eclipse.core.resource`s version 3.21.

You can check the version of the `org.eclipse.core.resources` plugin like this:
`Help -> About Eclipse Platform -> Installation Details -> Plug-ins -> type `org.eclipse.core.resources`:

[image]

Also, make sure you're using Eclipse Platform version 2025-03 (4.35), as it's required for the CDT LSP 3.x plugin:

[image]

If you need to update the Eclipse Platform, use the Help → Check for Updates option.

Mikl604, 2025-05-20:
Yes, it helped. Thanks!
But when opening main.c pops up a message like this:

[image]

Mikl604, 2025-05-20:
This only appears when the IDE is initially started and the main.c. of any project is opened for the first time. Then the projects open normally, and this message no longer pops up.

sigmaaa, 2025-05-20:
I'm glad to hear the initial issue has been resolved. We're now investigating the null pointer exception you reported. Since the original problem has been fixed, I'm closing this issue. Thank you for your input.

closed 2025-05-20 by sigmaaa