espressif/idf-eclipse-plugin #1232: ESP32-C5 is not in the launch target list

wqiu-calgary, 2025-05-30:
So I have installed the latest Espressif-IDE (Version: 3.4.0 Build id: 20250506-1512)
and the latest toolchain from the ESP-IDF master branch (ESP-IDF v5.5-dev-3951-ga74725a31b).

In the ESP-IDF command line, I was able to set esp32c5 as target with --preview as argument, and build it successfully. 

However, in the Espressif-IDE, esp32C5 is not listed as launch target. Is there a way to enable the preview features in Espressif-IDE?

wqiu-calgary, 2025-05-30:
adding a custom target does not resolve the issue. When creating a new launch target, it requires an IDF target, which is a drop down menu with a list of IDF supported targets, and ESP32-C5 is not in the list unfortunately. IDF support of ESP32-C5 is through --preview option

kolipakakondal, 2025-06-04:
Ok, good to know. Closing this issue, we will translate this into our offical docs as well for future reference.

Thanks

closed 2025-06-04 by kolipakakondal