espressif/idf-eclipse-plugin #1234: Unable to create an ESP32-P4 project. No ESP32-P4 target available in the latest 3.1.0 Espressif Eclipse release

velocitronics, 2025-06-04:
I need to use the ESP32-P4 chip and have noticed that there is support for it in Espressif 3.1.0. However, I am unable to select the ESP32-P4 target when creating an ESP32 project. When will a new Espressif Eclipse IDE be released that supports the ESP32-P4 chip? Is there a simple workaround to get something going? Please let me know.

Thanks,

Eric Norton

sigmaaa, 2025-06-05:
Hi Eric,

The ESP32-P4 target should be available by default in Espressif-IDE 3.1.0 and later. Which version of ESP-IDF are you currently using in the IDE? Please ensure it's v5.3 or newer, as support for the ESP32-P4 target requires at least that version.

velocitronics, 2025-06-05:
Hi Sigmaaa,

 I am using Espressif Eclipse IDE 3.1.0. When I create a project in the Espressif IDE, I do not get the option to select the ESP32-P4 chip as shown here:

[image]

Is there a workaround to this issue? I managed to install ESP-IDF 5.4.1, which allowed me to select the ESP32-P4 chip to create the project. However, I then encountered too many errors while trying to compile. I then said, 'Okay, well, this won't work,' so I uninstalled the Espressif IDE and reinstalled the 3.1.0 version. I tried to upgrade using the plugin for 3.4.0, I believe it was, but got a "clangd" error after trying to compile the ESP32-P4 project. When will a new version of the Espressif Eclipse IDE be released that doesn't have so many hurdles to climb over to use the ESP32-P4? I cannot use VSCode, and don't even get me started with Arduino. I like the ESP32-S3 as that does work, but severely limited in IO pins if using the RGB 565 interface, which I need to use because I cannot use the MIPI DSI interface, which would be nice, but the ESP32-S3 chip doesn't support it. I'm sure you can sense my frustration here.

Now I can compile the ESP32-P4 project fine. Upon trying to run a simple example:

/*
 * SPDX-FileCopyrightText: 2010-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

#include <stdio.h>
#include <inttypes.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_chip_info.h"
#include "esp_flash.h"
#include "esp_system.h"

void hello_task(void *pvParameter) {
    while (1) {
        printf("Hello world!\n");
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}

void app_main(void)
{
	xTaskCreate(hello_task, "HelloTask", 2048, NULL, 1, NULL);
}

 I get this error: 

E (33772) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (33772) task_wdt:  - IDLE0 (CPU 0)
E (33772) task_wdt: Tasks currently running:
E (33772) task_wdt: CPU 0: main
E (33772) task_wdt: CPU 1: IDLE1
E (33772) task_wdt: Print CPU 0 (current core) registers
Core  0 register dump:
MEPC    : 0x48153990  RA      : 0x480a77e4  SP      : 0x4ff30570  GP      : 0x4ff1b800
TP      : 0x4ff30760  T0      : 0xffff8000  T1      : 0x00000000  T2      : 0x00000753
S0/FP   : 0x00000001  S1      : 0x48641780  A0      : 0x48641768  A1      : 0x00000400
A2      : 0x00000006  A3      : 0x00001801  A4      : 0x4ff3058c  A5      : 0x00000003
A6      : 0x500a0000  A7      : 0x00000004  S2      : 0x4ff3f684  S3      : 0x00000003
S4      : 0x481de000  S5      : 0x486417a8  S6      : 0x00000004  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000019  T4      : 0x00000000  T5      : 0x0000005d  T6      : 0x00000781
MSTATUS : 0x00011888  MTVEC   : 0x4ff00003  MCAUSE  : 0xdeadc0de  MTVAL   : 0xdeadc0de
MHARTID : 0x00000000
Please enable CONFIG_ESP_SYSTEM_USE_FRAME_POINTER option to have a full backtrace.

I am sure there is a configuration somewhere I am missing, but not sure where to look. I just started trying to get something working with the ESP32-P4 and trying to get adjusted to the freeRTOS since I am a bare metal guy that doesn't like using an RTOS. I favor Nuvoton M480 chips over everything else any day of the week. They are easier to work with in my experience. The Espressif chips are nice too, but there is a steep learning curve to them.

I apologize for my frustration, but I don't have a lot of time to invest and was hoping I could make some headway here. Any advice would be helpful.

Thanks,

Eric Norton

velocitronics, 2025-06-05:
Hi Sigmaaa,

 After a grueling half of the day, I managed to get ESP-IDF 5.4.1 installed in the Espressif Eclipse 3.1.0. It was a lot more work than I wanted, but I got it done, and I can now compile and run ESP32-P4 code in Eclipse.

Here is what I did to get it to work:

CMakeLists.txt:

# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.16)

# Set the ESP-ROM ELF directory to prevent missing variable errors
set(ESP_ROM_ELF_DIR "C:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_rom/esp32p4/elf")

# Include the ESP-IDF build system
include($ENV{IDF_PATH}/tools/cmake/project.cmake)

# Define the project name
project(hello_world)

And...

gdbinit.cmake:

Line 40 -> #file(TO_CMAKE_PATH $ENV{ESP_ROM_ELF_DIR} ESP_ROM_ELF_DIR) <-- Comment this out
if(DEFINED ENV{ESP_ROM_ELF_DIR})
	set(ESP_ROM_ELF_DIR $ENV{ESP_ROM_ELF_DIR})
	file(TO_CMAKE_PATH "${ESP_ROM_ELF_DIR}" ESP_ROM_ELF_DIR)
else()
	message(WARNING "ESP_ROM_ELF_DIR is not set. Defaulting to empty path.")
	set(ESP_ROM_ELF_DIR "")
endif()

Hope this is useful for anyone in the same situation I was in.

Cheers,

Eric Norton

kolipakakondal, 2025-06-10:
Feel free to open a new issue if you encounter any other problems while working with the IDE. Closing this issue for now.