espressif/idf-eclipse-plugin #1253: In espressif ide clicking on mpr121.h opens it in bluefish

plater, 2025-06-18:
If I click on a representation or a function with prototype in mpr121.h then the c header file is opened in bluefish html editor - "file mpr121.h
mpr121.h: C source, ASCII text
[mpr121.h](https://github.com/nopnop2002/esp-idf-mpr121/blob/main/components/mpr121/mpr121.h)

plater, 2025-06-18:
Neither "File associations" nor "Content types" contain anything that would cause the mpr121.h to be associated with hypertext.

alirana01, 2025-10-08:
I am unable to reproduce this issue. If this persists kindly reopen the issue with more descriptive logs and operating system information

closed 2025-10-08 by alirana01

plater, 2025-10-08:
I've just rechecked, Eclipse Espressif-IDE Version: 3.4.0 Build id: 20250506-1512 and loading the project:
https://github.com/nopnop2002/esp-idf-mpr121 then opening the file "main/main.c" and holding the Ctrl key down while clicking on  "mpr121.h" on line 17 opens the file in bluefish ide.
This time it happened once but it now behaves correctly and opens the file in the standard editor.
It appears to have fixed itself.