espressif/idf-eclipse-plugin #1259: ESP-IDF Manager: 'Installing tools...' has encountered a problem. Cannot invoke "String.indexOf(int)" because "value" is null

rtek1000, 2025-06-29:
**Describe the bug**
A clear and concise description of what the bug is.

Related: https://github.com/espressif/idf-eclipse-plugin/issues/1040

![img](https://private-user-images.githubusercontent.com/529862/361907741-4842188f-fbe8-42c5-b927-ff454e35a10a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTEyMTcxMzgsIm5iZiI6MTc1MTIxNjgzOCwicGF0aCI6Ii81Mjk4NjIvMzYxOTA3NzQxLTQ4NDIxODhmLWZiZTgtNDJjNS1iOTI3LWZmNDU0ZTM1YTEwYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNjI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDYyOVQxNzA3MThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNDM2MmE1MTZhOTEwYTE4YmMzNjlmNTliZTI5Y2NhMGY5ZmI0MWJiNWRiODJiODUxMzcyMWEyMjBjYTUzMzY0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.1UK1DkZ5E-8gaYbvRmjMx7fLC00KUkOxzBK_pByWYWs)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

alirana01, 2025-06-30:
@rtek1000 please share the error logs and version you are using

rtek1000, 2025-06-30:
Hello, I downloaded it yesterday, it's the latest, Version: 3.1.0.

About the log, I'm afraid there isn't any more. I searched the Internet for how to find the log, but apparently it only has data about the compilation of the project.

After trying a few times to add the ESP-IDF (v5.4.2) that I had downloaded before the IDE (and it was working, I had already tested some examples, compiled on the command line), I tried to let the IDE download the ESP-IDF (v5.4.2) again (same version, it consumed much more mobile data than I was expecting), but the IDE managed to complete the installation.

The problem is in the option to use ESP-IDF which was already pre-installed, before the IDE. I installed ESP-IDF in the "~esp/esp-idf" folder (Linux).

sigmaaa, 2025-07-02:
Hi @rtek1000,

I responded in another ticket, but it applies here as well: the latest version is 3.5.0, available on the GitHub releases page: https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.5.0. I wasn't able to reproduce the error there.

rtek1000, 2025-07-02:
Hello, sorry, the problem is in the Espressif IDE, I just noticed that the repository is for IDE plugins. it's the latest, Version: 3.1.0.

https://developer.espressif.com/tags/espressif-ide/

closed 2025-07-02 by rtek1000