espressif/idf-eclipse-plugin #1260: Installation is downloading all files again

rtek1000, 2025-06-29:
**Describe the bug**
A clear and concise description of what the bug is.

After downloading large files and the installation finished with [errors](https://github.com/espressif/idf-eclipse-plugin/issues/1259). When I try to install again, the files are being downloaded again. They are large files, but I use mobile data. Why aren't the files already downloaded just being checked and reused?

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

rtek1000, 2025-06-30:
Hello, I downloaded it yesterday, it's the latest, Version: 3.1.0. (ESP-IDF v5.4.2)

About the log, I'm afraid there isn't any more. I searched the Internet for how to find the log, but apparently it only has data about the compilation of the project.

The problem also occurs if the installation is interrupted by the user, for example, if a file has already been downloaded and the user closes the IDE.

sigmaaa, 2025-07-01:
Hi @rtek1000,

The latest one is 3.5.0 available on the GitHub releases page: https://github.com/espressif/idf-eclipse-plugin/releases/tag/v3.5.0. I wasn't able to reproduce the error there

rtek1000, 2025-07-02:
Hello, sorry, the problem is in the Espressif IDE, I just noticed that the repository is for IDE plugins. it's the latest, Version: 3.1.0.

https://developer.espressif.com/tags/espressif-ide/