espressif/idf-eclipse-plugin #1284: UncheckedIOException

franz-ms-muc, 2025-08-06:
Get the Following Error: 

```
java.lang.RuntimeException: java.util.concurrent.CompletionException: java.io.UncheckedIOException: java.io.IOException: Cannot run program "/home/franz/.espressif/tools/esp-clang/esp-19.1.2_20250312/esp-clang/bin/clangd": error=2, Datei oder Verzeichnis nicht gefunden
java.lang.RuntimeException: java.util.concurrent.CompletionException: java.io.UncheckedIOException: java.io.IOException: Cannot run program "/home/franz/.espressif/tools/esp-clang/esp-19.1.2_20250312/esp-clang/bin/clangd": error=2, Datei oder Verzeichnis nicht gefunden
```

[image]

Version:
Espressif-IDE is an Integrated Development Environment(IDE) based on Eclipse CDT for developing IoT Applications using ESP-IDF framework.

Version: 3.5.0
Build id: 20250617-0727

[image]

sigmaaa, 2025-08-06:
> BTW: /home/franz/.espressif/tools/esp-clang/ there is nothing there.
> 
> but i did run the Setup like in the docs.

Hi @franz-ms-muc,

Normally, the .espressif/tools/esp-clang/ directory should be populated after a successful setup. Do you happen to have the logs from the tools installation? If not, could you try reinstalling the tools using the ESP-IDF Tools Manager in the IDE?

Once you've done that, feel free to share the installation log here if the issue persists — we’ll take a closer look

franz-ms-muc, 2025-08-14:
BUT: 

/home/franz/.espressif/tools/esp-clang/esp-19.1.2_20250312/esp-clang

now is existing ! 

tested with 
Espressif-IDE-3.6.0-linux.gtk.x86_64

closed 2025-08-14 by franz-ms-muc